Lora-net/LoRaMac-node #201: RX window offsets ignored in OTAA mode

desolatormkx, 2017-02-27:
Howdy,

I have ported the LoRaMAC software to my own custom hardware platform (EFM32 Jade Gecko based),  and have successfully sent/received packets in both EU and NA bands. 

The issue arises when using OTAA to commision the mote. Due to budget/size constraints my solution uses an internal RCO for RTC timing. This introduces a ~~bit~~ lot of slop in the RTC tick rate, that I can correct by offsetting my RX windows and increasing the symbol timeout. This solution works fine in ABP mode, however since the RX window timers are set by the network in OTAA these offsets are ignored, and downlink performance suffers.

I have added (and suggest adding) configuration parameter RX1_WINDOW_OFFSET and RX2_WINDOW_OFFSET that are applied to the rx window timers similarly to the RADIO_WAKEUP_TIME parameter.

Cheers,
Devin

mluis1, 2017-02-28:
You are free to add the RX1_WINDOW_OFFSET & RX2_WINDOW_OFFSET definitions to your specific platform adaptation.

We think that a time compensation/adjustment function is better suited for the implementation we provide.

Please note that for ClassB implementation a quite precise timing source will be required. Even using a XTAL source clock requires a compensation function for the ClassB implementation.

If you take a look at the ClassB branch you will find an example of such function.

desolatormkx, 2017-03-01:
Sounds good, I've decided to good with a hybrid solution. 
The offsets are there to account for fixed overhead in our design (SW IRQs, processing delays, etc).

To handle temperature/voltage related frequency changes I dynamically calculate a correction factor that uses the known Time-On-Air of a TX signal to indirectly calibrate the RCO against the SX1272s crystal. If more accuracy is needed in the future I'll explore the characterized temperature based time compensation as well. For now however, I'm able to receive on DR13 between ~-40 and +50 &deg;C 

Cheers

closed 2017-03-01 by desolatormkx