Lora-net/LoRaMac-node #204: overwrite the last data received

garyquan, 2017-03-02:
Hey, I have a question. Once the rx_done task is completed, the rx_done signal is triggered before entering the OnMacStateCheckTimerEvent, so that it seems to overwrite the last data received

mluis1, 2017-03-06:
Could you please elaborate a little bit more your question?
From your description we don't understand the issue.

- Please can you provide the steps on how can we reproduce the issue?
- Which branch are you using?
- Which LoRaWAN class are you using (A, C)?

garyquan, 2017-03-08:
1, the use of class C mode
2, the continuous transmission of multiple devices
3, I did not deal with the received data and then entered the OnRadioRxDone, resulting in the last data loss

Comments on the last sentence of the OnRadioRxDone function
Trig OnMacCheckTimerEvent call as soon as possible

This does not guarantee that you can not enter the OnMacCheckTimerEvent before entering the Rx_Done interrupt signal

mluis1, 2017-03-24:
We think that we have understood the issue.

We are investigating how to solve the issue and we will try to provide a fix as soon as possible.

mluis1, 2017-04-10:
Commit a32de9ac6c3f977b4e3ba30664c97d12ca8f1371 should solve this issue.

closed 2017-04-10 by mluis1