Lora-net/LoRaMac-node #205: Question about Join BACK-OFF

mpt8153, 2017-03-03:
In release v4.3.1  LoRaMacInitializationTime variable is initialized at every ResetMacParameters:

      static void ResetMacParameters( void )
      {

         // Initialize channel index.
        Channel = LORA_MAX_NB_CHANNELS;

        // Store the current initialization time
        LoRaMacInitializationTime = TimerGetCurrentTime( );
      }

This function is called  at every OnTxDelayedTimerEvent.

So with this modification I don't understand how back-off management is compliant to LoRaWAN specification. timeElapsed in JoinDutyCycle function is calculated in function of a timer that is updated continuosly and timeElapsed will be always  < 3600e3.

    static uint16_t JoinDutyCycle( void )
    {
       uint16_t dutyCycle = 0;
       TimerTime_t timeElapsed = TimerGetElapsedTime( LoRaMacInitializationTime );

       if( timeElapsed < 3600e3 )
       {
           dutyCycle = BACKOFF_DC_1_HOUR;
       }
       else if( timeElapsed < ( 3600e3 + 36000e3 ) )
       {
           dutyCycle = BACKOFF_DC_10_HOURS;
       }
       else
       {
           dutyCycle = BACKOFF_DC_24_HOURS;
       }
       return dutyCycle;

BR

djaeckle, 2017-03-06:
Hi mpt8153,

thanks for the report. Your analysis is correct. The initialization of the time is not at the correct place. We will fix the issue.

djaeckle, 2017-03-07:
Hi mpt8153,

fix is available on the develop branch. Also, I merged the develop with the feature/v4.4.0 and feature/classb branches.

GuillaumeEwattch, 2017-03-09:
Hi djaeckle,

it would seems to me that this fix would only work for the first join procedure, as the "LoRaMacInitialization" function is called only once.
I think that it would be better to reset the "LoRaMacInitializationTime" variable with each new join procedure. that would be in the "LoRaMacMlmeRequest" function.

kind regards

mluis1, 2017-03-09:
As defined by the LoRaWAN specification the BACK-OFF initial reference time is from end-device power-up or reset.

Please read Chapter 8 of 1.01 or Chapter 7 of 1.0.2 LoRaWAN specification.

GuillaumeEwattch, 2017-03-09:
Indeed,
Thank your answer.

closed 2017-03-17 by mluis1