Lora-net/LoRaMac-node #206: The "Radio" Global Variable, LoRaMac Modularity, and Multi-Radio Endpoints

loverdeg-ep, 2017-03-09:
This is a thought experiment as much as it is a request to deprecate the following global variable
`extern const struct Radio_s Radio;
`
Which is referenced across a number of sources in this repository.

It may just be a personal thing but the global Radio has always thrown a wrench in my ability to get a flow going when using LoRaMac.c and its supporting sources. 
Not wanting to request changes based purely on personal preference I'd like to make the following argument.

If for whatever reason one wanted to have two radio's in an endpoint managed by a single host controller, the LoRaWAN MAC as written would not support this.

`extern const struct Radio_s Radio;
`
radio.h declares it
The BSPs define and initialize it
LoRaMac.c uses it exclusively for radio management

Hosting a second radio would require creating a renamed copy of LoRaMac.c and it's non static APIs, in addition to any bsp functions LoRaMac.c relies on.

My personal preference says that LoRaMac should define an instance of a LoRaMac which contains a radio instance; But in a multi-radio scenario avoiding code duplication would require this.

The best example I could think of for needing multiple radios would be a class C endpoint that needs more down-link bandwidth.

Does anyone else see any advantage to eliminating the global "Radio" and improving the modularity of the LoRaWAN MAC?

tibogens, 2017-03-13:
+1, definitely. You don't even need an example, because [global variables are evil](http://wiki.c2.com/?GlobalVariablesAreBad)

mluis1, 2017-03-17:
@loverdeg-ep 
Basically your assessments are correct and make sense.
The issue is that we never thought about the usage of multiple radios.

In case you are willing to. You can always propose a solution that we will happily review and apply to the repository once agreed.

In the mean time I'll add to our todo list a task to try to remove the global variable dependency. Thus, allowing the usage of multiple radios.
Please note, that for the moment this task will get a low priority as there are more urgent topics to be addressed.

GuntherScuhlz, 2017-04-10:
@tibogens @tibogens @mluis1 

A quick-and-dirty is simply to refactor code to use a POINTER-to a struct Radio_s, dereferencing where neccesary. Then keep a second struct Radio_s for your alternate radio. Now just pass in the approriate pointer and job is done.

That's what I did in  my project, where I run 4+ physical radio chips