Lora-net/LoRaMac-node #208: Discuss about CN470-510MHz Band standard

ckt1010, 2017-03-13:
Hi all,
Hope there are more people care about CN470 Band. ^_^
My question is why CN470 band enable all 96 channel at beginning? And random all of them when it send message? 
How end point know which channel did gateway support? Did it means gateway should support all those channel at same time?

mluis1, 2017-03-17:
The CN470 band implementation enables all 96 channels by default because it is specified like that in the LoRaWAN specification.

The node connection process:
    * is to try all the 96 channels up until it gets an answer form the server.
    * As soon as the server receives an uplink from the node it can instruct the node to disable the unwanted channels.

In case you know which channels are to be used you can also instruct the node to use those at startup by disabling the unwanted channels. For this you may use the  following commands
```C
    uint16_t channelsMask[6];
                              // Default channels mask value
    channelsMask[0] = 0x00FF; // 0xFFFF;
    channelsMask[1] = 0x0000; // 0xFFFF;
    channelsMask[2] = 0x0000; // 0xFFFF;
    channelsMask[3] = 0x0000; // 0xFFFF;
    channelsMask[4] = 0x0000; // 0xFFFF;
    channelsMask[5] = 0x0000; // 0xFFFF;

    mibReq.Type = MIB_CHANNELS_DEFAULT_MASK;
    mibReq.Param.ChannelsDefaultMask = channelsMask;
    LoRaMacMibSetRequestConfirm( &mibReq );

    mibReq.Type = MIB_CHANNELS_MASK;
    mibReq.Param.ChannelsMask = channelsMask;
    LoRaMacMibSetRequestConfirm( &mibReq );

```

ckt1010, 2017-03-18:
Thanks for your code @mluis1 !
Actually I discuss why LoRaWAN specification specify it. Maybe I post wrong place....
As CN470 band can't use CFlist, even after I use your code, my end point will not move to other gateway...

mluis1, 2017-03-24:
Yes, you are right, it is not the right place to discuss the specification.

You may always try to send your requests/comments to the LoRa-Alliance help mail address. help@lora-alliance.org

closed 2017-03-24 by mluis1

ckt1010, 2017-03-25:
Thanks mluis1!