Lora-net/LoRaMac-node #209: GPS Rx buffer(NmeaString) run out of range!!!!

ckt1010, 2017-03-16:
Hi,
I see the NmeaString array size is 128, but why we use "if( ( data == '$' ) || ( NmeaStringSize >= 128 ) )" in GpsMcuIrqNotify?
If nmea packet size is equal to 128 last byte is '\n' it will cause NmeaString out of range.

mluis1, 2017-03-17:
Maybe I don't understand your concern but, an array in C is 0 based. This means that the 128th byte will be located at index 127. (The array index ranges from 0 to 127 which makes 128 in total)

I think that the implementation is correct as it prevents the "out of range" issue that you report.

ckt1010, 2017-03-18:
Hi @mluis1 ,
Sorry for confusion. I mean, when packet size is equal to 128 and last byte is '\n', you will see NmeaStringSize is 128 and data is '\n' after "NmeaString[NmeaStringSize++] = ( int8_t )data;".
And because data is '\n', so "NmeaString[NmeaStringSize] = '\0';" will be execute. But as NmeaStringSize is 128, so overflow will be happened.

mluis1, 2017-03-24:
Could you please try the following fix and let us know if you are ok with it.

We have changed `( NmeaStringSize >= 128 )` to `( NmeaStringSize >= 127 )`
and `NmeaString[NmeaStringSize] = '\0';` to `NmeaString[NmeaStringSize++] = '\0';`

````C
    if( id == UART_NOTIFY_RX )
    {
        if( UartGetChar( &Uart1, &data ) == 0 )
        {
            if( ( data == '$' ) || ( NmeaStringSize >= 127 ) )
            {
                NmeaStringSize = 0;
            }

            NmeaString[NmeaStringSize++] = ( int8_t )data;

            if( data == '\n' )
            {
                NmeaString[NmeaStringSize++] = '\0';
                GpsParseGpsData( ( int8_t* )NmeaString, NmeaStringSize );
                UartDeInit( &Uart1 );
                BlockLowPowerDuringTask ( false );
            }
        }
    }
```

ckt1010, 2017-03-25:
Thanks! I'll test it later.

closed 2017-04-10 by mluis1