Lora-net/LoRaMac-node #210: Can't clear LORAMAC_TX_RUNNING state after A UNCONFIRMED McpsRequest

twowinter, 2017-03-18:
LORAMAC_TX_RUNNING state should only be cleared by OnMacStateCheckTimerEvent with (LoRaMacFlags.Bits.MacDone == 1 );

And the MacDone can only be set by these Radio CallBack.

I am confused which CallBack can set MacDone  after A UNCONFIRMED McpsRequest.
 
[image]

twowinter, 2017-03-18:
### Is the MacDone flag setted by OnRxTimeout instead of OnTxDone

the RX2 is not start cause the default Receive window duration 3000ms, the RX2 RXtimeoutEvent can't be trigged.

After I modified the MAX_RX_WINDOW from 3000ms to 800ms, the RX2 go well, after RX2's OnRadioRxTimeout(), LORAMAC_TX_RUNNING state was cleared by OnMacStateCheckTimerEvent

#define MAX_RX_WINDOW                               800 // 3000

[image]

--------

Is LORAMAC_TX_RUNNING state should be cleared by RX2 RXtimeoutEvent? If yes, Is it need to change MAX_RX_WINDOW to avoid problem like me?

twowinter, 2017-03-20:
I tested a LoRaWAN BaseStation from CLAA(China Lora Application Alliance).
the RX1_WINDOW is should more than 1000ms, Is this a problem for  Gateway?

I should use RX1_WINDOW 3000ms in order to recvieve messages from gateway, and give up the RX2, it can't started.
------------
But this would cause my problem"Can't clear LORAMAC_TX_RUNNING state after A UNCONFIRMED McpsRequest", if the gateway didn,t answer to the end-device, it cann't send McpsRequest again.

mluis1, 2017-03-24:
I guess that you are misunderstanding how all this works.

The MAX_RX_WINDOW timeout should never happen in the reality. It is there just in case there is really something very wrong happening.

What defines the Rx windows timeouts is the `symbTimeout ` when the `Radio.SetRxConfig` function is called.

Once the `symbTimeout ` is set in the radio hardware and the radio is put in RxSingle mode. Then the Radio will automatically timeout after the number of specified `symbTimeout ` value.

In case the radio detects a preamble it is kept in reception mode up until the frame is fully received.

Please note that we made a commit to the develop branch that now computes the RxWindows parameters automatically.

closed 2017-03-24 by mluis1

twowinter, 2017-03-24:
thank you, maybe i found the reason.

My hardware didnot use DIO, i can only use REG_LR_IRQFLAGSMASK to judge RXDone/RXTimeout.

And the SX1276SetRx didnot set REG_LR_IRQFLAGSMASK with  RXDone/RXTimeout. So "very wrong happening".

eric780711, 2018-01-03:
Hi twowinter
I have the same problem, you have successfully solved it, the downlink problem?