Lora-net/LoRaMac-node #212: Can't setup Rx2 after Rx1error

ckt1010, 2017-03-20:
Hi,
Refer to issue #186 we can see rx1 receive delay will longer than rx2 window setup. So I see code change on commit d279bf3cd686f26852c437993c07d6b0f0b7a2a9.
But I think it still have problem: if rx1 window have crc error, then as OnRxWindow2TimerEvent will before than OnRadioRxError, and Rxslot will not set as 1 and rx2 window will not setup. And OnRadioRxError will handle nothing but set radio sleep. Then application layer can't receive any error status feedback, just waiting....

djaeckle, 2017-03-20:
Hi ckt1010,

yes, we also noticed this issue and have already a fix for it. Currently, it is under test. We will push it as soon as it has been verified.

ckt1010, 2017-03-21:
Thanks @djaeckle !

mluis1, 2017-04-10:
Commit d15d46e12fbeebc77684668165f35486943a3760 should solve this situation.

closed 2017-04-10 by mluis1