Lora-net/LoRaMac-node #214: MAC_TX_RUNNING in LoRaMacMcpsRequest

LaurentHB, 2017-03-24:
Hello,

I'm implementing some low power handlers and I'm currently having trouble with LoRaMacMCpsRequest returning LORAMAC_STATUS_BUSY because LoRaMAcState is MAC_TX_RUNNING and preventing me from sending a message to acknowledge a downlink from the gateway.

I'm basicallly letting my device enter sleep mode between the end of the RX windows and the beginning of TX. It stays in run mode between TX and RX for now.

In the OnRadioRxDone function, when I receive a confirmed down, I set a variable so that when I PrepareTxFrame() it will put a different than usual acknowledge packet. I also put my DeviceState back to DEVICE_STATE_SEND so that when it comes out of of OnRadioRxDone and through another cycle of the main loop, it will send the acknowledge packet right away ( Directly after receiving the confirmed down). 

Problem comes when I enter SendFrame() to send that acknowledge, it exists with LORAMAC_STATUS_BUSY because MAC_TX_RUNNING is the current state.

It doesn't seem to make sense, since the last thing my device was doing is receiving a confirmed down packet.

Earlier in my testing I had to comment these two lines in OnRadioRxDone since it would get stuck in an infinite loop of opening that event. It didn't seem to cause problems at the time when I wasn't going into low power and I wonder if it could be causing problems now.
//TimerSetValue( &MacStateCheckTimer, 1 );
//TimerStart( &MacStateCheckTimer );

Anybody ran into something similar? Thanks a lot!

srettAC, 2017-03-27:
Hi Laurent,

I had a similar problem a few weeks ago.
It was happening when I tried to join my network, when in fact the stack was still trying from my previous attempt.

Are you sure the previous message you tried to send, has been sent by the stack?
Maybe your device is still trying to join the network?
Or maybe you are sending confirmed uplinks, and the stack is still waiting for the acknowledgement?

LaurentHB, 2017-03-27:
Hello,

I'm sure the device joined the network, after the join the first message is sent and received by the gateway. I'm also sending unconfirmed uplinks so not really sure what else could be causing the issue.

The issue only happens when I receive a confirmed downlink from the gateway and try to send an answer right away.

LaurentHB, 2017-03-27:
Looked through the code some more to see how all of that is supposed to be working. Commenting 
//TimerSetValue( &MacStateCheckTimer, 1 );
//TimerStart( &MacStateCheckTimer );

definitely was a problem since the MacStateCheckTimer event is indeed what is used to clear the LoRaMacState . I uncommented it but it wouldn't enter  if( LoRaMacFlags.Bits.MacDone == 1 ) , I changed the timeout value from 1 to 50,  in case it didn't have time to be set properly before the event happened and it seemed to do the trick, it's now entering if( LoRaMacFlags.Bits.MacDone == 1 ) and presumably clearing the MAC_TX_RUNNING state.

It still seems to lock my execution some way because nothing happens after that and I keep going back through OnMacStateCheckTimerEvent.

LaurentHB, 2017-03-27:
Currently going through the MacStateCheckTimer to see how all of it works and see what might be different between timing out through OnRadioRxDone and OnRadioRxTimeout. 

In OnRadioRxTimeout, there is LoRaMacFlags.Bits.MacDone = 1;  the MacStateCheckTimer isn't started. It doesn't seem to have been started since it was last stopped after OnRadioTxDone.

In OnRadioRxDone , there is also LoRaMacFlags.Bits.MacDone = 1; but then right after the MacStateCheckTimer is started.  

I'm not sure I understand why there is that difference, anybody have some explanation for it?

Thanks!

kwmartin, 2017-03-28:
I'm guessing the reason for the difference is on OnRadioRxTimeout, the receive function is not finished. For example, it might still have to go through the RX2Window state. It continues until either successful or aborted and PrepareRxDoneAbort() is called which sets:
    LoRaMacFlags.Bits.McpsInd = 1;
    LoRaMacFlags.Bits.MacDone = 1;
and calls:
    TimerSetValue( &MacStateCheckTimer, 1 );
    TimerStart( &MacStateCheckTimer );

which calls OnMacStateCheckTimerEvent() after 1ms (i.e. almost immediately). Let me know if this helps as I haven't followed it through carefully.

LaurentHB, 2017-03-28:
Problem ended up being in my own implementation, something was accessing a timer that was not initiated properly and it made all timers crash so I would not get the OnMacStateCheckTimerEvent.  Thanks for the help!

closed 2017-03-28 by LaurentHB

kwmartin, 2017-03-29:
> Thanks for the answer @kwmartin <https://github.com/kwmartin>
>
> My system is fine when it goes through OnRadioRxTimeout. The problem
> comes when instead of OnRadioRxTimeout, my reception finishes with
> OnRadioRxDone after a CONFIRMED_DOWNLINK
>
> In that case, by the end of that function, my LoRaMacState is
> MAC_TX_RUNNING and even though these next 3 lines should clear the
> MAC_TX_RUNNING state by entering OnMacStateCheckTimerEvent, I never
> enter the event and my device doesn't not exit MAC_TX_RUNNING and
> locks up somewhere else because of my implementation.
>
> LoRaMacFlags.Bits.MacDone = 1;
>
> //Trig OnMacCheckTimerEvent call as soon as possible
> TimerSetValue( &MacStateCheckTimer, 1 );
> TimerStart( &MacStateCheckTimer );
>
> Is that clear? I'm not sure how to put so it makes sense!
>
> Thanks

Laurent, I don't follow why you don't get into 
OnMacStateCheckTimerEvent(), as the Timer should go off in 1ms. Could it 
be because interrupts are still disabled (OnRadioRxDone() is done under 
a DIO0 interrupt - the LoRaWan code does an awful lot of processing 
inside interrupts - the DIOx routines and the LoRaMac routines they 
call, and the timer service routines - and many of them are occurring 
asynchronously)?

-Ken

eric780711, 2018-01-04:
@LaurentHB Hi
What you're talking about is accessing a timer that does not start correctly, which will cause all the timers to crash, so I will not get OnMacStateCheckTimerEvent. Can you tell me which access department started the timer correctly?