Lora-net/LoRaMac-node #215: ADR not working when fBufferSize > 27

srettAC, 2017-03-27:
Hello,

I'm using this stack for an end device, and I have some issues with the ADR.

Everything work just fine when I send less than 28 Bytes in my payload.
However, when I send more than 27 Bytes (fBufferSize>=28), I noticed that the ADR is not working anymore. In fact, the board is not receiving any more downlink form my gateway.

This is happening for the default data rate DR_0. If I change it to DR_1, I can receive downlinks.

Since I'm using this stack on a different board, and had to rewrite some code (espescially the timers), I'm not sure if the problem comes from my implementation, or from the stack.
Is this a known bug?

Thanks!

PS: I noticed that for a fBufferSize>=28, the airTime is greater than 2s, maybe the problem comes from here?

djaeckle, 2017-04-05:
Hi srettAC,

thanks for the report. I could not reproduce the issue on EU868. Could you provide some more information please? Which branch (commit) are you using? Which class? Which region?

srettAC, 2017-04-05:
Hi,

I'm using the last master branch commit(release 4.3.1).
It's a class A device for EU863-870 frequency.

Thanks for watching

djaeckle, 2017-04-05:
Hi srettAC,

thanks for the feedback, I just checked out the master branch and on my side it is possible to send payloads >= 28 bytes. Also, the server is able to perform Link ADR Request commands.

The only change I did was to set `LORAWAN_APP_DATA_SIZE` to 50 bytes. Do you have any other hint for me to test your case?

The uplink air time should not be the problem, since the receive delays will be started at the end of an uplink.

GuntherScuhlz, 2017-04-05:
@srettAC  @djaeckle 

Hi guys.

Had **_exactly the same problem_** in earlier releases back in November 2016. (Could not receive packets whose payload was supposed to be 37 bytes or more).

After much toiling through the raw data, I discovered that it was **_NOT the mote_**, that was to blame, it was a **_gateway side error_**.

Specifically, it was with FASTNET as service provider, who's **_gateways_** specifically was **_generating the FCtrl.FOptsLen field incorrectly for payload >= 37 Bytes_** .

The causes the mote stack to run down the wrong path, and thus not decrypting sucessfully, and **appears** to be not getting downlink, when in fact it is receieving a **malformed downlink**, and thus discards it.

To confirm if this is your issue, on mote side, directly after "fCtrl.Value = payload[pktHeaderLen++];", insert "if(size>=(13+37)){ fCtrl.Bits.FOptsLen -= (size-((13+37)-1)); }".

Play with value <37> (both instances above), to determine exactly  what payload size is giving you issues.

If this is it, alert your WAN provider to fix their gateways PRONTO (as FASTNET did for me within 24hrs of me pointing out their bug)!

Hope this helps!

djaeckle, 2017-04-05:
Hi GuntherScuhlz,

thank you very much for sharing this information.

srettAC, 2017-04-05:
Thanks GuntherScuhlz,

I will try that, and let you know what I find.

srettAC, 2017-04-06:
Hi,

I tried to reproduce my problem on an other device we are developing, and I can't.
So it seems the problem was hardware releated, and not stack releated.

I'm sorry for the inconvenience.

closed 2017-04-06 by srettAC