Lora-net/LoRaMac-node #218: The TimerTime_t type is too small

temporaryaccount, 2017-04-05:
I'm working the MoteII board.

TimerTime_t is typedefed as uint32_t. In function RtcConvertCalendarTickToTimerTime you claim to support dates up to 2099. Considering that TimerTime_t holds values not in seconds but in STM32 RTC subseconds (which in current case is 1/2048 seconds) it can hold 0xFFFFFFFF / 2048 = 2097151 seconds maximum, which is about 24 days. Or maybe I am wrong? I can see other possible bugs in the rtc-board.c file if you are interested.

fletort, 2017-04-27:
+1 : i fall on this problem also today...... They are a limitation on the return value of the RtcConvertCalendarTickToTimerTime value that return a uint32_t.... and the tick reference is alwas year 00 (so 2000).... They are also a big limitation on this method, or maybe i am wrong too ?

jamesl-dm, 2017-05-24:
LoRaMacInitializationTime is bound to overflow too. I think the duty cycle timers may also overflow? Bumping the type up to 64 bits would probably fix it nicely?

temporaryaccount, 2017-05-24:
> Bumping the type up to 64 bits would probably fix it nicely?

Yes we ended up doing exactly that and other serious bugfixes in the rtc-board.c. If LoRa-Net is accepting (and eventually merging) pull-requests I can prepare one for the Mote II board.

nestorayuso, 2017-05-24:
Yes, I think rtc-board.c is a mess because i can't understand what variables or parameters are tick based and what are milisecond based. And why sometimes TimerTime_t is used and sometimes uint32_t is used.

Feel free to make a pull-request on the develop branch or say the public fork to look at.

temporaryaccount, 2017-05-24:
Please review the changes in [this branch](https://github.com/LocatorCJSC/LoRaMac-node/tree/mote_ii_rtc_refactoring
), and if it looks OK I'll create a pull request. Thanks.

temporaryaccount, 2017-05-25:
>  i can't understand what variables or parameters are tick (1/2048 seconds) based and what are millisecond based

We tried to fix that too in the another commit on the same branch.

We will open a pull request later today to move the discussion there if there is no objection.

hinaultd, 2017-06-05:
I have the problem, my programs stop after 49.7 days with the timertime_t in uint32_t.
The step of RTC is  488µs
488µs * 0xFFFF FFFFF =>  49, 7 days
The bug comes from the function "RtcConvertCalendarTickToTimerTime" which overflows, if  timertime_t in uint64_t the program stop approximately 65535 * 49, 7. "
Attention, check casts between functions and correct the functions "RtcGetElapsedAlarmTime"and "RtcComputeElapsedTime".

GregCris, 2017-12-21:
We have indeed found an issue with our implementation of the RTC on the STML0xx (MoteII board), the issue is located in the function "TimerTime_t RtcComputeElapsedTime( TimerTime_t eventInTime )" of the file rtc-board.c for the MoteII (\src\boards\MoteII\rtc-board.c).

The root cause was a mismatch in the rollover value due to the conversion from tick to ms.
The rollover of the computed value in tick actually occurs before the rollover of the value computed in ms and we were not handling this properly.

We will fix this issue as soon as possible in a coming release.
In the mean time, the issue can be fixed with replacing the RtcComputeElapsedTime as follow:

TimerTime_t RtcComputeElapsedTime( TimerTime_t eventInTime )
{
    TimerTime_t elapsedTime = 0;

    // Needed at boot, cannot compute with 0 or elapsed time will be equal to current time
    if( eventInTime == 0 )
    {
        return 0;
    }
    // first get the current value of the timer in tick
    elapsedTime = RtcConvertCalendarTickToTimerTime( NULL );
    // convert into ms
    elapsedTime = RtcConvertTickToMs( elapsedTime );

    // compare "eventInTime" with "elapsedTime" while watching for roll over due to 32-bit
    if( elapsedTime < eventInTime ) // // roll over of the counter
    {   // due to convertion tick to ms, roll over value is 0x7D000000 (0x7D000000 * 2.048 = 0xFFFFFFFF)
        return( elapsedTime + ( 0x7D000000 - eventInTime ) );
    }
    else
    {
        return( elapsedTime - eventInTime );
    }
}

We have been able to check carefully our RTC implementation with a new set of unit tests and the rollover of the 32-bit value is now fully functionnal.

mluis1, 2018-01-16:
We close the issue as a fix has been provided.

closed 2018-01-16 by mluis1