Lora-net/LoRaMac-node #219: Solving Issues with rtc-board for SensorNode board

smantonio, 2017-04-07:
While implementing the API for a STM32 custom processor. I've stumbled upon some issues which I consider I should share. Issues are:

-   Fix possible alarm elapsed time calculation error
    When using the function RtcGetElapsedAlarmTime() the time
    difference between the RtcCalendarContext and the current time
    is estimated with the rounded values of the timer values in ms.
    Estimating the difference based on rounded values can lead to 1ms
    differences to the actual elapsed time. This can cause the Timer
    module to miss the execution of a Timer Callback and get stalled.
    This commit allows calculation of the time difference based on the
    precise double value of the timer value, eliminating the possibility
    of this issue to occur.

-  Fix RtcComputeElapsedTime() for overflow cases
    When elapsedTime < eventInTime (overflow ocurred) the formula
    for correct calculation of the time difference is missing a
    unit increment.

-   Fix possible RTC alarm not attended
    This commit will ensure that the Alarm Subsecond configuration is 0.
    According to the compiler the contents alarmStructure may be different
    from 0, which can cause a subsecond alarm config.

[patch.txt](https://github.com/Lora-net/LoRaMac-node/files/906778/patch.txt)

fletort, 2017-04-27:
I thinks i got also the problem...
I often get the timer module get stalled, on the very small tcxo off event used on the I Cube Lrwan version.
Actually, i didn't find the source of the problem and i put a bigger time for this event.

How can i see that the timer module miss the executon of the callback ?

smantonio, 2017-04-27:
If you are using a debugger you can check if the Day/Hour/Minute/Second
configured in the RTC_ALRMAR register is behind the current
Day/Hour/Minute/Second hold in registers RTC_DR and RTC_TR. Even if you are
not using a debugger, you can manually check that by a manual condition on
the code.

The causes for this issue may be two:

   - ALRMASRR is configured to match subseconds
   - Issue with elapsed time calculation caused by double to in conversion.

You can check the patch I attached and apply the fixes and try them out.
Keep me posted! :)

heroichornet, 2017-11-23:
I have similar failure with the latest ST Stack -this still seems to be a problem.  The stack has the solution but it is commented out like bellow:

// S.R.: see for info https://github.com/Lora-net/LoRaMac-node/issues/219
// we simply add always one tick to make sure the time server always gets triggered
//return ((uint32_t)(1 + CalendarValue - RtcTimerContext.Rtc_Time));
  return ((uint32_t)(CalendarValue - RtcTimerContext.Rtc_Time));

I'm currently trying with the "1+" up until now it seems to be working.

fahad80, 2017-11-28:
@heroichornet  What ST stack version you are using? I don't have that commented-out part in my stack. I am using 1.1.2.

heroichornet, 2017-11-29:
@fahad80 I'm using 1.1.1 , I know its not the latest but I want to know why they have omitted the comment and change. Its strange , in 1.1.2 the comment has been deleted but the function code that gets compiled is the same. My question is, do I have to add +1 or not?

Regarding my comment above, the 1+ is probably not the root cause of my problem- still I need to know how it should be in the code.

heroichornet, 2017-11-29:
@fahad80  my appologies the comment comes from another developer in our team .... :sweat_smile:

mluis1, 2018-01-16:
Fixes for the current RTC have been provided on another issue #218.

We close this one.

closed 2018-01-16 by mluis1