Lora-net/LoRaMac-node #223: Downlink MAC command too long to reply

JiapengLi, 2017-04-28:
LoRaMac-node can only hanle uplink MAC command not greater than 15 ( **FOPTS** maximum size). 

But in some cases, server is possible to send downlink MAC command through port 0, this may make FOPTS too short to reply (consider multiple **NewChannelReq** commands, or in US915 multiple **LinkADRReq** commands to turn on/off channels). 

And it is even worse if consider the Maximum payload size, for example _US915_ DR0 frame can only carry 11 bytes payload (less than maximum 15 bytes).  If end-device (**US915** mode) receives too much MAC command and make MacCommandsBufferIndex bigger than 11, the stack is blocked (keep return `LORAMAC_STATUS_LENGTH_ERROR`)

My question is: 
+ Consider the payload limitation. Is it possible to reply one downlink MAC command by multiple uplink MAC commands? To split long MAC commands to pieces and send multiple times. (Not sure if it is easy to process for server)
+ Should LoRaWAN specification need clearly define MAX numbers of MAC commands which port 0 payload should contain?  When Downlink data rate  is higher than Uplink data rate, it is possible to make end-node (resources limited) difficult to reply  if server sends very long port 0 payload.

Thanks.

djaeckle, 2017-05-05:
Hi JiapengLi,

thanks for the report. This issue has been solved already and will be pushed soon. In case when the fOpts are greater than the maximum possible payload size, the node will drop the fOpts to enable the node to perform uplinks again. 

Regarding your questions:

- Currently, we consider a fragmentation of the MAC command answers as too much overhead.

- Its correct that this is a specification issue. You may contact the LoRa Alliance to provide your proposal.

JiapengLi, 2017-05-10:
@djaeckle 
Thank you for your explanation! I close this issue.

closed 2017-05-10 by JiapengLi