Lora-net/LoRaMac-node #226: Branch feature/v4.4.0 deleted

nestorayuso, 2017-05-05:
Today 11 commits were issued to feature/v4.4.0 branch, but the complete branch was deleted later.
The branch still can be seen at stackforce repository 
https://github.com/stackforce/LoRaMac-node/tree/feature/v4.4.0

djaeckle, 2017-05-05:
Hi nestorayuso,

thanks for the report. Today, we provided the update of the recent changes of the feature/v4.4.0 branch. Actually, we also merged the feature/v4.4.0 into the develop branch. This has been done according to the workflow defined in the wiki page: https://github.com/Lora-net/LoRaMac-node/wiki/LoRaMAC-node-Wiki.

The feature/v4.4.0 branch is no longer present, as we don't need it anymore. If you want to get the latest status of the LoRaWAN v1.0.2 implementation, you can checkout the develop branch and pull from the origin. 

Please also take a look at the wiki page for the roadmap, which provides an overview of our planned merges in the future: https://github.com/Lora-net/LoRaMac-node/wiki/LoRaMAC-node-Roadmap.

nestorayuso, 2017-05-05:
I see the wiki updated, thanks.

closing...

closed 2017-05-05 by nestorayuso