Lora-net/LoRaMac-node #227: Development branch 4.4

coljander, 2017-05-11:
Hi

We're keen to get working on the 4.4 code, however on briefly looking at the code it appears there have been major changes re the events and how packets are sent.  

Specifically the file that handled this functionality, LoRaMac-api-v3.c, doesn't appear to exist anymore.  It's even been removed in the example project.

Is it a case this will be added when the conformance testing has been completed so we just wait or is this all the actual code that will be pushed to the master branch and we have to add this in ourselves (probably by re-incorporating LoRaMAc-api-v4.c?  We don't want to start making all the necessary changes if the code in the development branch is going to substantially change.

Thanks

pdemil, 2017-05-16:
See https://github.com/Lora-net/LoRaMac-node/wiki

Release Version 4.3.1

Release date:

27.02.2017

Contents:

US915 compliance according to LoRaWAN Specification v1.0.1.
**Stop support for v3 wrapper implementation.**

coljander, 2017-05-16:
Is there a specific reason for this?

mluis1, 2017-05-19:
The intention for LoRaMac-api-v3 was to let enough time to the users to move to the new API.

In version 4.3.0 we announced that we would deprecate the v3 API at version 5.0.0. But, in the mean time we decided to remove the support for it at version 4.3.1.

The reason for this was that supporting different APIs would require more work in order to get them synchronized and in plus the API v3 doesn't support all the new features that have been added to the new API.

closed 2017-06-19 by mluis1