Lora-net/LoRaMac-node #228: different crystal use

batulzii, 2017-05-14:
please, advice to port if i use different crystal in Mote. We have design based Loramote but with different crystal frequency.

djaeckle, 2017-05-16:
Hi batulzii,

could you please provide some more information? Which crystal do you want to change?

batulzii, 2017-05-16:
we use 147456Mhz crystal for stm32L1516.  Where in which file (HAL) need to changes to keep right timing.

djaeckle, 2017-05-16:
Do you mean the STM32L151R6? If not please provide the related datasheet.

batulzii, 2017-05-16:
Oh, sorry for short part number. it is a STM32L151CBT6

djaeckle, 2017-05-16:
Table 29. denotes that a maximum frequency of 24MHz is supported for the HSE. Or do I look in the wrong datasheet? Please provide the link to the datasheet to avoid missunderstandings.

batulzii, 2017-05-16:
ok. 14Mhz is fine for stm32L151CB . question is where and in which file need to edit .
i had made changes as bellow:

#if !defined  (HSE_VALUE) 
  #define HSE_VALUE    ((uint32_t)8000000)

 changed to

#if !defined  (HSE_VALUE) 
  #define HSE_VALUE    ((uint32_t)14745600)

djaeckle, 2017-05-17:
Hi batulzii,

the definitions of the clock frequencies are defined in file `stm32l1xx_hal_config.h`. You have to change the define for `HSE_VALUE`. You could either define `HSE_VALUE` in your preprocessor options or edit it in the file directly.

closed 2017-05-17 by batulzii