Lora-net/LoRaMac-node #229: Class C OTAA Join RX2 Problem

gimasi, 2017-05-14:
We have been experiencing an issue not being able to Join with a Class C node on a network that responds to join on RX2.
We have spotted the problem in the LoRaMac.c on OnRadioRxTimeout function, this is how currently it is setup:
static void OnRadioRxTimeout( void )
{
    if( LoRaMacDeviceClass != CLASS_C )
    {
        Radio.Sleep( );
    }
    else
    {
        OnRxWindow2TimerEvent( );
    }

    if( RxSlot == 1 )
    {
        if( NodeAckRequested == true )
        {
            McpsConfirm.Status = LORAMAC_EVENT_INFO_STATUS_RX2_TIMEOUT;
        }
        MlmeConfirm.Status = LORAMAC_EVENT_INFO_STATUS_RX2_TIMEOUT;
        LoRaMacFlags.Bits.MacDone = 1;
    }
}

You need on the if ( RxSlot == 1 ).... a check on the Class C, so it should be like this:

 if(( RxSlot == 1 ) && ( LoRaMacDeviceClass != CLASS_C ))
...
...

During the Class C join, after TXDone an OnRxWindow2TimerEvent is fired. No RX arrives so the RxTimeout is entered, and being RxSlot=1 the MacDone bit is sent.  A MacStateCheckTimer event can come in ( usually every second ) and in the OnMacStateCheckTimerEvent function, having MacDone = 1 we get into the 'MlmeConfirm.MlmeRequest == MLME_JOIN' case, and a ReJoin is started... this before the RX2 has time to reach the preamble time and receive the Join Ack.

Hope I was clear and that other people can experiment it.

Massimo

djaeckle, 2017-05-15:
Hi Massimo,

thanks for the report. I'm already analyzing it. Does the network respond in the first RX2 or in the second RX2 window?

gimasi, 2017-05-15:
It responds on the second RX2 window.
Massimo

batulzii, 2017-05-16:
Massimo, Thank you! We had problem OTAA with C class

MisterQH, 2017-05-22:
Hi Massimo, I was experiencing the same issue with Class C OTAA. Thank you !

gimasi, 2017-05-22:
I'm happy it helped!

djaeckle, 2017-05-24:
Hi Massimo,

I was able to reproduce the issue. We will provide an update soon.

djaeckle, 2017-05-30:
Hi Massimo,

i have just updated the develop branch. Please take a look at commit 34a8fbb1d5.

alexanderplanar, 2017-06-16:
Yep! It works now! I had been trying to make OTAA works with my LoRaMote-Kit from Semtech for long time. Now after fixing the issue, all works!

closed 2017-06-19 by mluis1