Lora-net/LoRaMac-node #231: Function to return max payload size for next TX

ivandigiusto, 2017-05-18:
Hi,

I am looking for a function that will return the max size of payload for the next TX.

All I could find is **LoRaMacQueryTxPossible( uint8_t size, LoRaMacTxInfo_t\* txInfo )**
which gives you True/False answer if you feed it a number of bytes that you are trying to transmit.

In case of an application which can send a variable payload, it would be extremely useful to have a function that returns the max payload size for the next TX.  This number can vary from TX to TX due to changes in:
- ADR
- SF
- FOpts

and may need to be checked before each transmission.

Alternatively, we could use the existing LoRaMacQueryTxPossible function, but we would have to call this function from size=1 all the way to a size that returns False.  Not very efficient.

Any chance of getting this added in version 4.4?  The function (e.g. **LoRaMacNextTxMaxPayload**) would just be a refactoring of **LoRaMacQueryTxPossible** function.

Regards,
Ivan

djaeckle, 2017-05-18:
Hi ivandigiusto,

you are looking at the correct function. Please take a closer look at the parameter `LoRaMacTxInfo_t* txInfo `. The function `LoRaMacQueryTxPossible` will fill this parameter with the information you requested.

The structure is defined as:
```
/*!
 * LoRaMAC tx information
 */
typedef struct sLoRaMacTxInfo
{
    /*!
     * Defines the size of the applicative payload which can be processed
     */
    uint8_t MaxPossiblePayload;
    /*!
     * The current payload size, dependent on the current datarate
     */
    uint8_t CurrentPayloadSize;
}LoRaMacTxInfo_t;
```
The structure contains `MaxPossiblePayload`, which holds the maximum payload which you can send in the next uplink. The field `CurrentPayloadSize` provides the maximum possible payload on the next datarate. In general:

`MaxPossiblePayload = CurrentPayloadSize - fOpts`

The function takes already ADR into account.

ivandigiusto, 2017-05-19:
I see now that the function does two things:
* respond with a OK / Not OK
* update the parameters of max payload size

We can definitely use the function as is.

Thanks for the quick reply!

closed 2017-05-19 by mluis1