Lora-net/LoRaMac-node #233: Function to test when the next transmission can be requested (EU MAC/duty cycle)

catsup, 2017-05-19:
Hello,

I have a buffering mechanism in place to queue up several frames and submit them to the MAC for transmission when a transmission is possible. Ideally, I'd need a function to let me know when to wake up for sending the next buffered frame, ie, at what point the MAC will be done with its current tx/rx sequence and - where applicable - any duty cycle limitations.

Is there a function to test when the next transmission can be requested, or plans to add one ?

Thanks,

svenKautlenbach, 2017-05-20:
Not exactly what you are requesting for, but currently as far as I know one can use

> uint8_t LoRaMacLinkCheckReq( void );

from LoRaMac-api-v3.h for newer API check the implementation of the same very function. You may setup periodical setup of the link.

jamesl-dm, 2017-05-24:
+1: I also need a mechanism like this. When sending status updates, you'd like to know when the update will be stalled for several minutes waiting for a duty cycle backoff. That way you can send a fresh status update when the wait is over, rather than a queued update from 10 minutes ago.

GuntherSchulz, 2017-08-24:
Hi @catsup @djaeckle 

see my comments on #233

mluis1, 2018-01-16:
Will be solved once issue #296 fix is finalized.

closed 2018-01-16 by mluis1