Lora-net/LoRaMac-node #234: LoRaMac.c's AggregatedTimeOff is never reset

jamesl-dm, 2017-05-24:
I haven't tested this personally, but it looks like AggregatedTimeOff is never reset. So if a gateway sets the node duty cycle, AggregatedTimeOff will increase with every transmission, causing larger and larger delays. 

Originally, it was reset to zero during the channel selection procedure, once the LastAggrTx time got large enough for a transmission. But it looks like this code got dropped when the channel selection code was factored out of LoRaMac.c.

djaeckle, 2017-05-24:
Hi jamesl-dm,

thanks for the report. I will check it.

djaeckle, 2017-05-30:
Hi jamesl-dm,

I have updated the develop branch. Please take a look at 59a21a5e.

jamesl-dm, 2017-05-31:
Thanks @djaeckle, looks like that'll do it

closed 2017-05-31 by jamesl-dm