Lora-net/LoRaMac-node #235: RegionAU915 maintains a set of unused channels, but doesn't use it

jamesl-dm, 2017-05-24:
At 
[https://github.com/Lora-net/LoRaMac-node/blob/develop/src/mac/region/RegionAU915.c#L771](url), CountNbOfEnabledChannels is called using ChannelsMask, instead of the carefully maintained ChannelsMaskRemaining:
```
if( nextChanParams->AggrTimeOff <= TimerGetElapsedTime( nextChanParams->LastAggrTx ) )
    {
        // Search how many channels are enabled
        nbEnabledChannels = CountNbOfEnabledChannels( nextChanParams->Datarate,
                                                      ChannelsMask, Channels,
                                                      Bands, enabledChannels, &delayTx );
    }
```

So channels can be reused many times in a row, which seems to be what ChannelsMaskRemaining was intended to prevent.

djaeckle, 2017-05-24:
Hi jamesl-dm,

thanks for the report. I guess your statement is correct. I will double check it and will come back to you soon.

djaeckle, 2017-05-30:
Hi jamesl-dm,

I have just updated the develop branch. Please take a look at 1f35250.

closed 2017-05-31 by jamesl-dm