Lora-net/LoRaMac-node #238: nextTxDelay not assigned in RegionAU915NextChannel - stalls tx

jamesl-dm, 2017-05-24:
At [RegionAU915.c#L770](https://github.com/Lora-net/LoRaMac-node/blob/6e01e908e76722bb77772fb1466a685771749181/src/mac/region/RegionAU915.c#L770), nextTxDelay is not assigned. In other regions, there is a call to RegionCommonUpdateBandTimeOff that sets nextTxDelay, but in AU915 it remains -1 == UINT32_MAX. This leads to transmissions stalling indefinitely.

djaeckle, 2017-05-24:
Hi jamesl-dm,

thanks for the report. I think the issue is that I missed to update the initialization value of `nextTxDelay` for the US based bands. I will double check it on my test system and will provide feedback asap.

djaeckle, 2017-05-30:
Hi jamesl-dm,

I have just updated the develop branch. Please take a look at d95ab107.

jamesl-dm, 2017-05-31:
Hi djaeckle,

I believe you need something like:
```
if( nextChanParams->AggrTimeOff <= TimerGetElapsedTime( nextChanParams->LastAggrTx ) )
{
    // Reset Aggregated time off
    *aggregatedTimeOff = 0;

    // Update bands Time OFF
    nextTxDelay = RegionCommonUpdateBandTimeOff( nextChanParams->DutyCycleEnabled, Bands, AU915_MAX_NB_BANDS );

    // Search how many channels are enabled
    nbEnabledChannels = CountNbOfEnabledChannels( nextChanParams->Datarate,
                                                  ChannelsMaskRemaining, Channels,
                                                  Bands, enabledChannels, &delayTx );
}
```

Without the call to RegionCommonUpdateBandTimeOff, nextTxDelay will be zero, even if no channels were available due to the join duty cycle. This causes an immediate transmission, with *channel uninitialized.

djaeckle, 2017-05-31:
Hi jamesl-dm,

thanks for the feedback. I will review it and come back to you soon.

closed 2017-07-07 by mluis1