Lora-net/LoRaMac-node #239: AU915's LimitTxPower is still using US915 figures

jamesl-dm, 2017-05-24:
At [RegionAU915.c#L88](https://github.com/Lora-net/LoRaMac-node/blob/6e01e908e76722bb77772fb1466a685771749181/src/mac/region/RegionAU915.c#L88), the LimitTxPower has not been changed from the US915 implementation.

The figures from the latest LoRaWAN regional parameters are 30 dBm for 125 kHz, and 26 dBm for 500 kHz channels.

I'm also not sure why the band definitions seem to have the default power inserted instead of the max power ([example](https://github.com/Lora-net/LoRaMac-node/blob/6e01e908e76722bb77772fb1466a685771749181/src/mac/region/RegionAU915.h#L172))?

djaeckle, 2017-05-30:
Hi jamesl-dm,

I have just updated the develop branch. Please take a look at dc87d5e6 and 75591a699.

jamesl-dm, 2017-05-31:
I think the MAX should be a MIN as well?

```
// Limit tx power to the band max
txPowerResult =  MAX( txPower, maxBandTxPower );
```

djaeckle, 2017-05-31:
Hi jamesl-dm,

thanks for your feedback. I think the implementation is correct. Please note that the TX power is inverse, which means that a txPower of 0 (`TX_POWER_0`) provides the highest output power. 

Assume we want to limit the maxBandTxPower to `TX_POWER_2` and txPower is `TX_POWER_0`. Then we we have to limit it to `TX_POWER_2`, which is the MAX.

jamesl-dm, 2017-05-31:
Ah - missed that. Thanks!

closed 2017-05-31 by jamesl-dm