Lora-net/LoRaMac-node #240: ping-slot-time

qingchuwudi, 2017-05-24:
Hi, 

How does it work fine with different time units in function `CalcNextSlotTime`? 

```C
static bool CalcNextSlotTime( ... ... )
{
   ... ...
   /***** 
    * currentTime, BeaconCtx.LastBeaconRx and BeaconCtx.Cfg.Interval is in the unit of second.
    * So slotTime is in the unit of second, too.
    *****/
    slotTime = ( ( currentTime - BeaconCtx.LastBeaconRx ) % BeaconCtx.Cfg.Interval );
    slotTime = currentTime - slotTime;
    slotTime += BeaconCtx.Cfg.Reserved;
    
    /***** 
    * PingSlotWindow is 30 milliseconds. 
    * What about "slotTime + slotOffset * PingSlotCtx.Cfg.PingSlotWindow"? 
    * How does it work?
    *****/
    slotTime += slotOffset * PingSlotCtx.Cfg.PingSlotWindow;

    ... ...
    /***** And *****/
        slotTime += ( ( TimerTime_t )( currentPingSlot * pingPeriod ) *
                    PingSlotCtx.Cfg.PingSlotWindow );
    ... ...
}
```
I doubt about it.

Branch [feature/class_b](https://github.com/Lora-net/LoRaMac-node/tree/feature/class_b)
_ _ _

djaeckle, 2017-05-24:
Hi qingchuwudi,

thanks for the report. Everthing is calculated in milliseconds. `BeaconCtx.LastBeaconRx` and `BeaconCtx.Cfg.Interval` are both in milliseconds. Here are some details about the function:

We are interested in the time when we should have received the last beacon. In the first code line, we subtract the last beacon rx time from the current time. The result is a time delta. In case we have missed a beacon, this delta could possibly be greater than the beacon interval. Therefore, we perform the modulo operation. In the second code line, we subtract the time delta from the current time.
```
// Calculate the point in time of the last beacon even if we missed it
slotTime = ( ( currentTime - BeaconCtx.LastBeaconRx ) % BeaconCtx.Cfg.Interval );
slotTime = currentTime - slotTime;
```
Now, we know when the last beacon rx should have happend. In the next step, we add the beacon reserved time and the ping slot offset.
```
// Add the reserved time and the ping offset
slotTime += BeaconCtx.Cfg.Reserved;
slotTime += slotOffset * PingSlotCtx.Cfg.PingSlotWindow;
```
The value of `slotTime` contains now the time stamp of the first ping slot window. If the `currentTime` is greater than this time, it means we have to open up the next ping slot window. If not, we will open up the first ping slot window. In case we are opening up a ping slot window which is greater than `currentTime`, we are calculating the time of the next ping slot window as follows:

```
if( slotTime < currentTime ) { // Next ping slot is not the first one, calculate the next one
// We are calculating the number of the upcoming ping slot
currentPingSlot = ( ( currentTime - slotTime ) /
( pingPeriod * PingSlotCtx.Cfg.PingSlotWindow ) ) + 1;
// We are calculation the time when we have to open the next ping slot
slotTime += ( ( TimerTime_t )( currentPingSlot * pingPeriod ) *
PingSlotCtx.Cfg.PingSlotWindow );
}
```
In class b, you have 4096 ping slots. The variable `pingPeriod ` is the time between ping slots expressed in number of slots. In the first code line, we compute the delta between the current time and the first ping slot window. We devide it by `pingPeriod * PingSlotCtx.Cfg.PingSlotWindow` which is the time between two ping slots in milliseconds. E.g. if the `currentTime` is between ping slot 2 and 3, you should get a value between 2 and 3, which will be stripped-down to 2. Therefore, we add +1, since we want to have the next ping slot, namely ping slot 3.
In the second step we are just multiplying this number with the ping period [slots] and the ping slot window [ms]. The result should be the trigger time for the next ping slot.

qingchuwudi, 2017-05-25:
@djaeckle Thanks very much.

closed 2017-05-25 by qingchuwudi