Lora-net/LoRaMac-node #244: JoinAccept message not received when AS923_DEFAULT_DOWNLINK_DWELL_TIME is set to 1

anthonyblx, 2017-05-31:
I have noticed that RegionAS923RxConfig() function will calculate the maxPayload based on the DownlinkDwellTime. If the DownlinkDwellTime is 1 then 
maxPayload = MaxPayloadOfDatarateDwell1DownAS923[dr];

Which is a problem when we receive the JoinAccept message on the RxWindow1. 
As a result, Radio.SetMaxPayloadLength( modem, maxPayload + LORA_MAC_FRMPAYLOAD_OVERHEAD ); will set the maximum payload to 24 (maxPayload = 11, LORA_MAC_FRMPAYLOAD_OVERHEAD  = 13).

This is smaller than the JoinAccept which is 34 bytes long. 

Wouldn't it be better to ignore the DownlinkDwellTime  when we are waiting on JoinAccept?

Regards,
Anthony

anthonyblx, 2017-06-01:
Here is a fix I am proposing:
In OnRxWindow1TimerEvent():
`if (IsLoRaMacNetworkJoined == false)
    {
    	RxWindow1Config.DownlinkDwellTime = 0;
    }
    else
    {
    	RxWindow1Config.DownlinkDwellTime = LoRaMacParams.DownlinkDwellTime;
    }`

gimasi, 2017-06-01:
Anthony,
I have noticed the same problem - I fixed it temporarily forcing maxpayload to 51 - but I don't think its the right solution - yours look better !
Massimo

calebbramwell, 2017-06-05:
We ran into the same issue a while back. 
As the only effect the DownlinkDwellTime has is how long we listen for, it would seem as though there would be no problem in setting the `AS923_DEFAULT_DOWNLINK_DWELL_TIME` to 0 and leaving it up to the network server to set it if relevant. 
This method allows the NS to send a MAC payload larger than 11 bytes in the first downlink slot etc.

mluis1, 2017-07-07:
Solved with commit 66b81e5a1d61f3873f1e93ddc318be2d0342ac4c

closed 2017-07-07 by mluis1