Lora-net/LoRaMac-node #248: Class-B multiple join-request error

TuDong, 2017-06-06:
Subsequent program blocking,if the first join-request has been failure.

### First Join-Request
```C
LoRaMacStatus_t LoRaMacMlmeRequest( MlmeReq_t *mlmeRequest )
{
   ... ...
    else
    {
        MlmeConfirmQueueCnt++;
    }
   ... ...
}
```
 The first join-request,`MlmeConfirmQueueCnt++ `.

### First Join-Request Failure ,and send join request again.
```C
static void OnTxDelayedTimerEvent( void )
{
   ... ...
  if( ( LoRaMacFlags.Bits.MlmeReq == 1 ) && ( index < LORA_MAC_MLME_CONFIRM_QUEUE_LEN ) ) 
    {
        ResetMacParameters( );
        ... ...
    }
  ... ...
}
```
In the function 'ResetMacParameters()','MlmeConfirmQueueCnt=0',and send another join request.

### program blocking

- No data received.
```C
static void OnMacStateCheckTimerEvent( void )
{
  ... ... 
  if( ( LoRaMacFlags.Bits.MlmeReq == 1 ) || ( ( LoRaMacFlags.Bits.McpsReq == 1 ) ) )
   {
        index = GetMlmeConfirmIndex( MlmeConfirmQueue, MLME_JOIN, MlmeConfirmQueueCnt );
         if( ( LoRaMacFlags.Bits.MlmeReq == 1 ) &&
                   ( index < LORA_MAC_MLME_CONFIRM_QUEUE_LEN ) ) 
         {
               ... ...
         }
         ... ...
  }
  ... ...
}
```
Because there is no operation on MlmeConfirmQueueCnt.The program hangs.

- Receve data.
```C
static void OnRadioRxDone( uint8_t *payload, uint16_t size, int16_t rssi, int8_t snr )
{
  ... ... 
  index = GetMlmeConfirmIndex( MlmeConfirmQueue, MLME_JOIN, MlmeConfirmQueueCnt );
  if( index < LORA_MAC_MLME_CONFIRM_QUEUE_LEN )
  {
     ... ...
   }
  ... ...
}
```
The same problem as above.

TuDong, 2017-06-16:
should you have any further  suggestions?

djaeckle, 2017-06-19:
Hi TuDong,

thanks for your request. I will double check it.

djaeckle, 2017-06-29:
Hi TuDong,

I was able to reproduce the issue. We will provide a fix soon.

TuDong, 2017-06-29:
@djaeckle Hi djaeckle,
Thank you very much for your attention.

djaeckle, 2017-08-08:
Hi TuDong,

thanks again for the finding. I will push an update for feature/class_b soon.

djaeckle, 2017-08-08:
Hi TuDong,

we have pushed a new verison to the feature/class_b branch. It contains a fix for your issue. Please let me know if it is resolved. Please note that you have to define the symbol `LORAMAC_CLASSB_ENABLED` to enable class b related code parts in the sources now.

mluis1, 2018-01-16:
The fix was provide on commit 21b3f07b1be051cf9630330fe16f717b6a45e764

closed 2018-01-16 by mluis1