Lora-net/LoRaMac-node #249: rxTimeOut Problems

m4tts, 2017-06-06:
i'm currently developing a lora application on the murata lora-chip. everything works fine, until I take the power off my gateway. for example when i want to join, or when a message with ack doesn't get acknowledged. i get an rxTimeOut, and then the LoRaMac does something strange with the timers...
it keeps setting them longer and longer and the stack got stuck. does anyone has the same problem?

djaeckle, 2017-06-19:
Hi m4tts,

thanks for your report. Wich version are you using currently? You can specify the number of trials for the join request. If the node doesn't get an answer or the answer is not valid, it will try to join again. What do you mean by "long timeouts"? Which region are you using?

m4tts, 2017-06-19:
i was using the B-LRWAN library from ST but switched the loramac to the version available at from semtech.com (V4.10). there is nbr of trials no longer supported. this version runs stable in our system.
i'm operating the node in switzerland, so the region is EU868.

If i would set the Nbr of trials to 3, what happens if the number of trials equals 3? will i get a notification that joining didn't work? i couldn't figure it out in the code by myself...

djaeckle, 2017-06-23:
Hi m4tts,

I would recommend to clone/download the versions which we provide at GitHub. Here, we will always provide the most recent versions (currently, the latest version is 4.3.2). In this version, the number of trials is supported for the join procedure.

You can either clone the repository and check out any version or you could download it here: https://github.com/Lora-net/LoRaMac-node/releases.

closed 2017-07-07 by mluis1