Lora-net/LoRaMac-node #250: Rx2Channel.Datarate never takes into account

ygaklk, 2017-06-07:
Hi,

I noticed a strange behavior on this line: 
https://github.com/Lora-net/LoRaMac-node/blob/develop/src/mac/LoRaMac.c#L1411
Why the Rx2_frequency is updated, but not the Rx2_Datarate ?
(By adding this line for example: RxWindow2Config.Frequency = LoRaMacParams.Rx2Channel.Frequency; )

Because, during a MacCommand "RxParamSetupReq", Datarate and Frequency for Rx2 are updated, but the datarate will never be used during the next Rx2 windows.
Code here: https://github.com/Lora-net/LoRaMac-node/blob/develop/src/mac/LoRaMac.c#L1720

Thank you

djaeckle, 2017-06-19:
Hi ygaklk,

thanks for your report. The MAC updates the datarate in function `RegionComputeRxWindowParameters` which redirects to the specific implementations of the related regions.

ygaklk, 2017-06-23:
Hi djaeckle,
Thank you for your reply. And sorry, my question was not clear.

I agree with the use of `RegionComputeRxWindowParameters` during a Tx, to compute the Rx2 windows parameters.
But my case is the following: I produce a MIB_set_request to change the datarate of the Rx2Windows, and then I turn my device in classC. At this point, the device is listening on the wrong datarate (because no Tx was performed).

This problem is visible in classC only, my first post was not clear enough.

djaeckle, 2017-06-23:
Hi ygaklk,

thanks for the feedback. I will take a look at this issue and will come back to you.

fahad80, 2017-06-28:
I have similar kind of problem (in Class C). Right now I am doing a dummy Tx  :-|

djaeckle, 2017-08-08:
Hi ygaklk,

we have applied a fix for it. Is the issue solved with it?

ygaklk, 2017-08-23:
Hi djaeckle & mluis1,

I've tested this [patch](https://github.com/Lora-net/LoRaMac-node/commit/3197008ae46cdc3ee62ab02c8d08f1a29491b358), and I think that the fix is good, but it is not usable in this state.

The first line of the patch checks if the device is already in classC. But, I think, in most of cases, the application layer wants to configure the Rx2_freq + Rx2_Dr before turning the mac layer into classC (from classA state). This is not possible with this patch, because the request will be discarded due to the classA state of the device.

My question is: why do you want to check the device class before applying the rx2 params ?
Because I think that the app_layer can change the Rx2 params even in a classA device (and it is the responsibility to the device to indicate this change to the LNS, in order to be coherent)

fahad80, 2017-09-01:
Ok, this is the problem I faced. When I tried to change the Rx2 window parameters, it throws "LORAMAC_STATUS_PARAMETER_INVALID" from line 2809 (LoRaMac.c). In other words, RegionRxConfig returning false, that's because the Radio is not in RF_IDLE mode (it is in RF_RX_RUNNING mode), due to being in Class C.

Adding this line fixed my problem:
`if( ( LoRaMacDeviceClass == CLASS_C ) && ( IsLoRaMacNetworkJoined == true ) )
 {
                     Radio.Sleep( );   // <------------------This line
                    // Compute Rx2 windows parameters
                    RegionComputeRxWindowParameters( LoRaMacRegion,
                    LoRaMacParams.Rx2Channel.Datarate,
                    LoRaMacParams.MinRxSymbols,
                    LoRaMacParams.SystemMaxRxError,
                    &RxWindow2Config );`

djaeckle, 2017-09-06:
Hi ygaklk, 

thanks for the feedback. 

> The first line of the patch checks if the device is already in classC. But, I think, in most of cases, the application layer wants to configure the Rx2_freq + Rx2_Dr before turning the mac layer [...]

I wasn't aware of this use case, but I think it seems correct. We will discuss it and come back soon.

djaeckle, 2017-10-09:
Hi all,

this issue is on the roadmap for the next release:

https://github.com/Lora-net/LoRaMac-node/wiki/LoRaMAC-node-Roadmap

closed 2018-01-16 by mluis1