Lora-net/LoRaMac-node #251: TX datarate used after join accept

sda3422, 2017-06-08:
Hi, this is a question but depending on your answer maybe it could lead to an optimization in the stack.
When ADR is activated and after a successful activation, why do not use this datarate for the first frame sent (and so use this datarate as start point for ADR) ? Today, the datarate used at first frame sending is always the default (SF12), ADR activated or not.
When ADR is not activated it is up to the application to choose its datarate so I agree that it is not linked to your implementation. Thanks !

nestorayuso, 2017-06-08:
I think, when ADR is activated, it is up to the network server to choose node's datarate.
If network server after the join, loads to the node new channels or parameters via MAC commands, the node will start with the right data rate. 

The election of the datarate is done by the network server. the node can just decrease (but not increase) the datarate if there is no downlink traffic

sda3422, 2017-06-08:
The TX datarate is not included in join accept, this means that the first frame will always be sent in DR0 if this datarate is the default one in the product. By using the rx window after this first emission the server will be able to change the datarate if ADR is activated. But, it could be more optimized to immediately use higher datarate at first TX if the join request has been received and accepted. It also allows to save downlinks especially if the ADR management is not very efficient on server side (going from DR0 to DR5 in more than on step).

sda3422, 2017-06-12:
In `OnRadioRxDone`, `case FRAME_TYPE_JOIN_ACCEPT:` treatment, the current datarate is reset to default one : `LoRaMacParams.ChannelsDatarate = LoRaMacParamsDefaults.ChannelsDatarate;`.
The result is the description I made before when ADR is activated but also it is impossible to retrieve the datarate even if ADR is not activated by using in `MlmeConfirm`:
`mibReq.Type = MIB_CHANNELS_DATARATE;
LoRaMacMibGetRequestConfirm( &mibReq );`
So, still the same remark : it could be valuable to inform the application on join request accepted datarate...

djaeckle, 2017-06-19:
Hi sda3422,

thanks for your report. The LoRaWAN Specification does not define the TX datarate after the activation. Currently, the LoRaMac uses the default one. 

Do I understand it correctly, that  your proposition would be to inform the application layer of the TX datarate of the previous join request after a successful join?

sda3422, 2017-06-22:
@djaeckle, Hi ! There are 2 points, the 1st is when ADR is not activated : yes, TX datarate of successful join should be provided to the application. 
2nd point, when ADR is activated : TX datarate is handled by the the stack itself and so, first emission should be executed on the TX datarate of successful join. Of course, nothing has to change on the default TX datarate value.

djaeckle, 2017-06-23:
Hi sda3422,

1. We will discuss this internally.
2. When the ADR bit is set, the Server is responsible to optimize the datarate. Usually, before the Server starts the optimization, it first collects some statistics about the connection quality. We think it is a better way to let the Server handle the datarate completely.

djaeckle, 2017-10-09:
Hi sda3422,

this issue is on the roadmap for the next release:

https://github.com/Lora-net/LoRaMac-node/wiki/LoRaMAC-node-Roadmap

closed 2018-01-16 by mluis1