Lora-net/LoRaMac-node #253: should the join accept CFList delete all preconfigured channels?

GuillaumeEwattch, 2017-06-15:
Hello,
In the lorawan specification 1.0.1 chapter 7.1.4 regarding EU863-870 JoinAccept CFList
this sentence can be found:
"If present, the CFList replaces all the previous channels stored in the end-­device apart from the three default channels as defined in Chapter 7."

Doesn't this mean that all previously configured channels of the end device should be deleted before applying the CFList channels?

The difference being that channels beyond channel 8 are currently not affected by the CFList and are staying configured and used after the join accept. This leading to potential packet loss if the gateway is not condigured to listen to these channels.

Regards,
Guillaume

mluis1, 2017-06-23:
Thanks for the report. We will update the MAC implementation to reset all channels upon receiving the JoinAccept message.

GuillaumeEwattch, 2017-06-23:
happy to help.

closed 2017-06-23 by GuillaumeEwattch