Lora-net/LoRaMac-node #254: adc-board.c AdcMcuReadChannel() shuts down HSI

ChrisEAlfred, 2017-06-20:
Why do all implementations of adc-board.c AdcMcuReadChannel() contain:
// Disable HSI
__HAL_RCC_HSI_DISABLE( );
at the end?

This is called via PrepareTxFrame(). Isn't the processor running the HSI at this point? In which case, the HSI is prematurely turned off by the read of the battery calling AdcMcuReadChannel().

It seems quite bizarre (wrong?) to be managing the CPU clock from an ADC read!

mluis1, 2017-06-20:
If you take a look to the `void SystemClockConfig( void )` in board.c file you will notice that only HSE and LSE oscillators are used.

But, the ADC peripheral requires the usage of the HSI oscillator. Thus, the oscillator is enabled when performing the ADC reading and stopped at the end as it is no more required.

Looking at the stm32l1xx_hal_adc.c file you will notice the following sentence inside the comments (shown below)
> *Only one possible clock source: derived from HSI RC 16MHz oscillator (HSI).* 
ADC is connected directly to HSI RC 16MHz oscillator.

```c
/*
...
        (++) Two clock settings are mandatory: 
             (+++) ADC clock (core clock).
             (+++) ADC clock (conversions clock).
                   Only one possible clock source: derived from HSI RC 16MHz oscillator
                   (HSI). 
                   ADC is connected directly to HSI RC 16MHz oscillator.
                   Therefore, RCC PLL setting has no impact on ADC.
                   PLL can be disabled (".PLL.PLLState = RCC_PLL_NONE") or 
                   enabled with HSI16 as clock source
                   (".PLL.PLLSource = RCC_PLLSOURCE_HSI") to be used as device 
                   main clock source SYSCLK.
                   The only mandatory setting is ".HSIState = RCC_HSI_ON"

             (+++) Example:
                   Into HAL_ADC_MspInit() (recommended code location) or with
                   other device clock parameters configuration:
               (+++) __HAL_RCC_ADC1_CLK_ENABLE();

...
*/
```

ChrisEAlfred, 2017-06-20:
So all the ports use HSE and nothing else can use HSI (without changing adc driver).

Is it a pre-requisite to use HSE (i.e. very accurate) for the LoRaMac-node?

mluis1, 2017-06-23:
We have used HSE because we had it on all our platforms.

You are free to use HSI if your platform doesn't have it. 
The `AdcMcuReadChannel( )` is implemented on a file that is specific to your platform. 
Thus, you are free to adapt all the files under `boards\<platform name>` to your specific platform needs. The only obligation is to respect the public APIs.

closed 2017-07-07 by mluis1