Lora-net/LoRaMac-node #255: Is SX1276GetTimeOnAir() rerun us in FSK

twowinter, 2017-06-21:
Hello,

there is no details of 'time on air' in FSK mode in the datasheet, so i doubt the return value in FSK.

GuntherScuhlz, 2017-06-21:
See section 4.25 in the datasheet - all equations for times to get from sleep to Tx. When in Tx, time on air is dictated by (data rate x [total number of bits sent, including preamble, sync, payload, crc]). See also table 42 for equations in the register map table, giving you actual results based on register settings. The code DOES follow these faithfully

GuntherScuhlz, 2017-06-21:
@mluis1 and other devs. Related to this topic, I've been meaning to mention forever, but never get got around to it, the following:
The use of floating point calcs is not only evil in terms of code resources for size-limited microprocessors, but also entirely unneccesary. Since every single register is an integer value, all calcs can and and have integer derivations and results. I always rewrite my copy of the stack to ban all floating points, so there is zero loss of accuracy due to rounding etc. The only errors are truncation errors, which is a DEVICE limitation, since they have integer registers.

Maybe after 4.4.0?

mluis1, 2017-06-21:
@GuntherScuhlz Yes, you are right. This is something that is currently in the tasks pipe.

If you are willing please feel free to propose your solution to this.

GuntherScuhlz, 2017-06-21:
Hecticly busy at work these next two days, but tell me which fork you'd like me to redo equations on - should be up by early next week.

My way to pay it forward for your guys excellent work

GuntherScuhlz, 2017-06-21:
p.s. this would include updating reference tables used in code taken straight from the datasheet, where even those tables are only approximations to true values gotten from equations in datasheet - eg: const FskBandwidth_t FskBandwidths[] follows Table 22 exactly, but actual value used from registers result in
const FskBandwidth_t FskBandwidths[] =
{
    { 2604  , 0x17 },
    { 3125  , 0x0F },
    { 3906  , 0x07 },
    { 5208  , 0x16 },
    { 6250  , 0x0E },
    { 7812  , 0x06 },
    { 10416 , 0x15 },
    { 12500 , 0x0D },
    { 15625 , 0x05 },
    { 20833 , 0x14 },
    { 25000 , 0x0C },
    { 31250 , 0x04 },
    { 41666 , 0x13 },
    { 50000 , 0x0B },
    { 62500 , 0x03 },
    { 83333 , 0x12 },
    { 100000, 0x0A },
    { 125000, 0x02 },
    { 166666, 0x11 },
    { 200000, 0x09 },
    { 250000, 0x01 },
    { 333333, 0x00 }, /* Invalid Bandwidth */
};

closed 2017-07-07 by mluis1