Lora-net/LoRaMac-node #256: Re-Join (JOINREQ_NBTRIALS) - How does this work

gcgasson, 2017-06-21:
Hi,

I am using the CMWX1ZZABZ-091 from Murata and using the ST I-CUBE-LRWAN lib.

Everything works great but I have found the following problem and just would like to know how to handle it.

The units will keep on communicating to the Multitech Gateway with no problem but once force the gateway to re-new the lease so that all the units will need to re-connect again. But once I do this the units will not re-connect again unless I recycle the power and then the units will join the gateway again.

Is this correct or should the units re-join automatically again?

mluis1, 2017-06-23:
The end-devices have no way to know if the server has renewed the leases.

This is something that must be handled at the application layer.
Before renewing the leases the server must first inform the end-device that this will happen. Then the end-device can start sending JoinReq again.

pjb-loxius, 2017-06-27:
Hi mluis1,

This means the server must be fail-safe/redundant. Otherwise a power failure would render all end-devices without proper measures in their application useless .....

RakshitAdmar, 2017-07-31:
@pjb-loxius 
The network server persists the leased parameters. It is expected that the server doesn't lose this on power cycle.

closed 2018-01-16 by mluis1