Lora-net/LoRaMac-node #258: just send one packet!

zhoubx123, 2017-06-23:
HI .
     Why does my module send only one packet of data, and then no more data?. The same is true if the electricity is re energized.
     Debugging found that the program did not sleep again , I never stopped at the RTC_Alarm_IRQHandler (void) setting breakpoint, except for the two time it started.

djaeckle, 2017-06-23:
Hi zhoubx123,

based on the information provided it is really hard to help you. Could you please provide the version, the platform, possible modification and so on for your case?

zhoubx123, 2017-06-24:
I am using master release/v4.3.2 , and the project is LoRaMac-node-master\Keil\SensorNode\LoRaMac\classA. I want to test 470MHz, so I changed 433MHz to
#define LC1                { 470500000, { ( ( DR_5 << 4 ) | DR_0 ) }, 0 }
#define LC2                { 470700000, { ( ( DR_5 << 4 ) | DR_0 ) }, 0 }
#define LC3                { 470900000, { ( ( DR_5 << 4 ) | DR_0 ) }, 0 }
and my mcu is stm32l151CCU6, so STM32L151xC and startup_stm32l151xc.s used.

zhoubx123, 2017-06-25:
After sending the first packet of data, it should go into sleep mode, but debugging has found that it has been unable to go into sleep mode since it sent the first packet of data.

fahad80, 2017-06-28:
When you say sleep mode, did you mean the Low_power_mode or the Lora state DEVICE_STATE_SLEEP?

RakshitAdmar, 2017-07-31:
It could also be sending data on channels that you haven't programmed to listen on the gateway. Having an SDR really helps.

mluis1, 2018-01-16:
No more activity we close the issue.

closed 2018-01-16 by mluis1