Lora-net/LoRaMac-node #259: timer.c TimerLowPowerHandler() HasLoopedThroughMain

ChrisEAlfred, 2017-06-23:
What is this code for and why '5'?

`        if( HasLoopedThroughMain < 5 )
        {
            HasLoopedThroughMain++;
        }
        else
        {
            HasLoopedThroughMain = 0;
            if( GetBoardPowerSource( ) == BATTERY_POWER )
            {
                RtcEnterLowPowerStopMode( );
            }
        }
`

mluis1, 2017-06-23:
The why '5' it was arbitrary chosen.

The goal of this is just to be sure that all events have been processed before going into sleep.

ChrisEAlfred, 2017-06-27:
The comment for HasLoopedThroughMain states:
"This flag is used to make sure we have looped through the main several time to avoid race issues"

What are the races?
Seems rather critical to be arbitrary. What is the minimum value of HasLoopedThroughMain?

If we modify the example code, how do we know to change HasLoopedThroughMain?

mluis1, 2017-06-29:
I agree that the comment is misleading. We will change it to something like the following:
>This flag is used to loop through the main several times in order to be sure that all pending events have been processed.

The number of loops isn't important (or critical). The goal of this is to cope with the situation where an event arrives just before going to sleep. By looping through main at least 1 time we ensure that this event is processed before going into sleep.

closed 2017-07-07 by mluis1