Lora-net/LoRaMac-node #261: RTC

rohitkumar5293, 2017-07-05:
Hey hi,

i am porting this semtech end device code for my custom MCU. Everything is fine but my MCU won't have RTC crystal so how can i use RTC for next transmission of packet and RTC alarm to wake my MCU from sleep after completion of duty cycle..

Reply ASAP

Regards
Rohit

tibogens, 2017-07-05:
> i am porting this semtech end device code for my custom MCU

What MCU? I guess by custom you mean "board"

> Everything is fine

Please define "everything"

> my MCU won't have RTC crystal

The LoRa stack need to be driven by an accurate real-time clock. You might get away without a crystal for your MCU but this sounds like asking for trouble. For having carried out tests using the MCU's internal oscillator, it won't work except for dumb TX in ABP.

> how can i use RTC to [...]

This info can be RTFMed out of your "custom MCU" manual.

> Reply ASAP

Done. You are welcome.

rohitkumar5293, 2017-07-05:
Thanks for reply....

What MCU? I guess by custom you mean "board"
 - Yeah here i mean board.

Please define "everything"
- here i mean i can able to get through the code but having only problem regarding RTC.

- What i mean to ask is - our lora packet will consist time stamp also while transmission? 
- If End Device won't have RTC but lora gateway can do time stamp if lowaran protocol is used? if it is possible then how lora gateway can do time stamp for END device.

Regards 
Rohit

mluis1, 2018-01-16:
No more activity we close the issue.

closed 2018-01-16 by mluis1