Lora-net/LoRaMac-node #262: When will v4.4 (LoRaWAN 1.0.2) code will be released?

ivandigiusto, 2017-07-06:
Hi,

We are trying to understand when v4.4 release is going to happen?

The main reason is so that we can benefit from new LoRaWAN 1.0.2 MAC commands (and all the bug fixes).

The other reason is that a lot of product manufacturers are unable to send us AS923 devices as they are waiting for Semtech's v4.4 code.

Some visibility on the timelines would be appreciated, as LoRaWAN 1.0.2 has been out for a while, and LoRaWAN 1.1 has still ways to go before getting approved.

Thank you,
Ivan

mluis1, 2017-07-07:
On an implementation point of view we think that we are ready to perform the release.

But, in order to perform a release we need to have at least 2 test-houses confirming that all certification tests are passed.

We are currently preparing the firmware images to be sent to the test houses. Please note that the certification tests take quite long to run.

We hope to be able to do the 4.4 release in the coming 2 to 3 weeks if everything goes smoothly.

tibogens, 2017-07-07:
Congrats and good luck with the certification! Thanks

ivandigiusto, 2017-07-07:
Thanks Miguel, that is great news.

closed 2017-07-26 by ivandigiusto