Lora-net/LoRaMac-node #263: Floating point computations removal

mluis1, 2017-07-07:
Hi GuntherSchulz,

As discussed on Issue #255 I have just created the feature branch "feature/floating-point-removal".
Please feel free to propose your changes to this branch.

Thank you very much for your proposal.

Best regards,

Miguel

GuntherSchulz, 2017-07-09:
@mluis1 

Hi Miguel,

Push is coming in two phases: Added new compiler (now), and then the actual floating-point cleanup (later this week).

1. Added my preferred compiler, IAR Embedded Workbench for ARM, v8.11.1.13272 to the repository. 

2.Made two seperate workspaces, one with ALL WARNINGS OFF, so code compiles cleanly as-is. Second one with ALL WARNINGS ON [OMG there is some cleanup to do on this build]. 

3.Also some minor code clean-up - ensuring all USE_BAND_xx semantics are replaced with REGION_yy equivalent [please specifically check definition of LORAWAN_APP_DATA_SIZE for the regions I added - comment is clear], fixing printf's in assert_failed, and making sure all assert_failed are identical.

4. Actual float fixes coming later this week, please check the push of 1-3 for now

FML took me 3 friggin days to put all the projects together - but that was the hard part, now the easy part (fixing floats) begins!

Best Regards,
Gunther

GuntherSchulz, 2017-07-09:
@mluis1 
Yo man, can't push the first update (need permission) - can you help please?
The sooner the better, so changes in 3. above ripple through system before other changes

Ty

Gunther

GuntherSchulz, 2017-07-10:
@mluis1 
Hi. I see you're active at the moment. If you can help with the above, I can hopefully cranks out the floating point fix today still

GuntherSchulz, 2017-07-10:
@mluis1 
Finally RTFMed it and now Grokking! 1-2 above in pull request. 3 to follow now. 4 later today/tomorrow

GuntherSchulz, 2017-07-21:
@mluis1 
This was finally completed about 6 days ago - see Feature/floating point removal #267

mluis1, 2017-07-21:
@GuntherSchulz 
Sorry for the late feedback. We are currently quite busy preparing the 4.4.0 release.

We would like to thank you for the great work provided.

Please find below our first comments.

1. The first goal was only to provide the changes concerning the floating point computations removal. But, actually there are different topics covered by your pull-request.
2. It would be nice if you could split the big pull-request in smaller ones and by topic. Currently we can't accept the pull request as is.
3. We will need some more time to validate that all proposed changes are OK. We hope to integrate them once we agreed upon for the 4.4.1 release.
4. Some of the updates that you have done are also done on our internal repositories. We didn't had the time yet to push them on GitHub. Mainly it concerns the example applications update.
5. Concerning the IAR IDE support, we are sorry but we can't merge these changes. The first reason is because we have no IAR license thus, we can't verify the projects. The second reason is that we would like to reduce the number of supported IDEs in order to reduce the amount of time spent to update all the different projects for the different targets at every release.
Mainly our goal is to remove the coIDE and Keil projects support and replace it by a CMAKE based building system. The CMAKE building system has the advantage to be portable across the different PC operating systems while coIDE, Keil and IAR aren't. 
The issue is that we can't make every one happy as every one has his own preferences.

GuntherSchulz, 2017-07-21:
@mluis1 
Thanks for the feedback, much appreciated

1. Extra stuff was just because it needed doing :)
2. Can you not add individual updates? They are explicitly tagged by topic in pull request. Floating-point only changes are:
       Radio Subdirectory: 2d703f0 + c73c26d   (no prerequisites)
       mac subdirectory:    ad3d41a                    (only requires change in utilities.h from 2d703f0)
       delay.c and delay.h: 32103fc                      (only requires change in utilities.h from 2d703f0)
       mpl5115:                 9112cc0                      (only requires change in utilities.h from 2d703f0)
       gps:                         bf46eee + 3f6501d     (only requires change in utilities.h from 2d703f0)
       board.c:                   f1507d7                      (only requires change in utilities.h from 2d703f0)
       rtc:                           e16ca09                      (only requires change in utilities.h from 2d703f0)
3. Great
4. Presumably
       update assertion failure:                           435101c
       USE_BAND_xxx replced by REGION_yyy:  3364082
       Chip Errata:                                               204f2eb + ff450da
5.  No worries man, if FEEL THE PAIN on managing so many projects. +1 for going CMAKE

If there's any way I can help more, let me know :)

mluis1, 2017-07-21:
@GuntherSchulz 

2. As far as I'm aware off we can only accept/refuse the full pull-request changes. That's the reason why I asked if it was possible for you to split the propositions. We can do it by our self but then your contribution will not be seen.

4. Yes, for the 2 first changes.
Concerning the errata we need to verify your proposition as we were convinced that we implemented all the required errata changes.

GuntherSchulz, 2017-07-21:
@mluis1 
2. Don't need the glory - if its easier for you to do it, go ahead. I you WANT me to do it, i'm happy to - your call
4. Sure

mluis1, 2017-07-21:
@GuntherSchulz 
2. Ok, I will do it and mention your name in the different commits.

loverdeg-ep, 2017-07-21:
+2 for cmake

GuntherSchulz, 2017-08-25:
@mluis1 

I did see the 2 commits you made, but of course the meat and bones of floating point removal is still to be merged. Presumably 4.4.0 certification process is keeping you guys quite busy.

Btw updating the Compliance Test Status on the Roadmap page (hasn't updated in ages) would help people with judging release date more accurately (I know a lot of guys are waiting on not deploying solutions to public just because it SHOULD be soon).

Keep up the good work, and as always, thanks for your collective monumental efforts! :)

mluis1, 2018-01-16:
This task is on our internal issue tracker. It has low priority for the moment.
Thus, we close it as well as the pull-requests #267 and #272.

closed 2018-01-16 by mluis1