Lora-net/LoRaMac-node #266: Is it possible to switch from ClassA to ClassB on the fly

mluis1, 2017-07-10:
Question asked on issue #217 by @griphi.

> Hi, I have a question regarding to the class system of the LoRaWAN.
> Is it possible to program the MCU to switch from class A to class B after an specific interrupt?
> 
> Example:
> 80% of the time the MCU is in class A mode (sleep mode) and after an hardware interrupt the MCU switches into the Class B mode (send Beacon).
> 
> Hardware:
> STM32 P-Nucleo-LRWAN1 with SX1272
> Gateway Conduit from Multitech with LoRaWAN Card extension
> 
> Thanks

mluis1, 2017-07-10:
Yes, the firmware can be switched to any class (A/B or C) on the fly. 

The only constraint with class B is that once you request the switch to class B the end device will try to acquire the beacon before really switch to ClassB.

griphi, 2017-07-10:
Thank you for your reply.

Currently, I am using "End Node" for class A communication.
How can I switch to any class?

djaeckle, 2017-08-08:
Hi griphi,

the develop branch contains sample applications for class A and C. A sample application for class B is present on the feature/class_b branch. Please note that the end node doesn't send beacons. It only receives beacons.

closed 2018-01-16 by mluis1