Lora-net/LoRaMac-node #268: RX timeouts when OTA joining

temporaryaccount, 2017-07-10:
Sometimes my node is having issues joining the network, I am getting RX timeouts, so I inserted some debug strings with timestamps (in milliseconds) when the node was trying to join:

```
154 try tx
169 OnRadioTxDone
5089 try rx(3000)
5123 OnRadioRxTimeout
6013 try rx(3000)
6221 OnRadioRxTimeout
6594 try tx
6670 OnRadioTxDone
11586 try rx(3000)
11620 OnRadioRxTimeout
12508 try rx(3000)
12716 OnRadioRxTimeout
13247 try tx
13323 OnRadioTxDone
18241 try rx(3000)
18276 OnRadioRxTimeout
19165 try rx(3000)
19373 OnRadioRxTimeout
19905 try tx
19982 OnRadioTxDone
24901 try rx(3000)
24936 OnRadioRxTimeout
25825 try rx(3000)
26033 OnRadioRxTimeout
26567 try tx
26644 OnRadioTxDone
31551 try rx(3000)
31586 OnRadioRxTimeout
32473 try rx(3000)
32681 OnRadioRxTimeout
33212 try tx
33288 OnRadioTxDone
38195 try rx(3000)
38230 OnRadioRxTimeout
39116 try rx(3000)
...
...
```

Seeing those small timeouts I assume my problem comes from the fact that I use the internal LSI clock for the RTC. LSI is not very accurate, and joining the network seems to require some precise timings. Can this be the reason?

I also noticed that RX1 timeout window is much smaller than RX2 timeout window (see the debug strings above, i.e. 34ms vs 208ms).

The specification says:

> The length of thereceive window is five symbols long and therefore depends on the data rate being used. It is the minimum time required to effectively detect a downlink preamble.

So I'd also like to understand whether this difference is deliberate (for example the reason may be that RX2 is using SF12 while RX1 uses SF7) or is a result of some kind of a bug.

Thanks

fahad80, 2017-07-10:
LSI won't work. I tried it. This requires precise timing which LSI can't provide.

temporaryaccount, 2017-07-11:
Thanks, that answers the first part of the question then.

mluis1, 2017-07-25:
The Rx timeout difference between Rx windows is due to the different datarates in use. 

5 symbols at SF12 doesn't give the same time as 5 symbols at SF7.

temporaryaccount, 2017-07-26:
Thanks @mluis1, I am closing the issue.

closed 2017-07-26 by temporaryaccount