Lora-net/LoRaMac-node #269: How can i use this library base on the other MCU

talahata, 2017-07-11:
This project used STM32L151C8 but i don't want to use this microcontroller. My team is using an other MCU and it is MKL16z128. It's an ARM Cotex M0 chip. 
What changes should I make to my project so that I can use this library?

Thanks!

temporaryaccount, 2017-07-11:
At least you will have to implement a "board", see this [directory](https://github.com/Lora-net/LoRaMac-node/tree/master/src/boards) for implemented boards. At least you'll have to implement the board.c, gpio-board.c, rtc-board.c, spi-board.c and sx1272-board.c (or sx1276-board.c) modules. The other main parts of library should be cross-platform as far as I know, at least on ARM Cortex-M.

finnoliver, 2017-07-24:
Documentation is horrible x_x

djaeckle, 2017-08-08:
Hi all,

we will provide an update of the documentation with the upcoming release. It will contain a chapter containing a short porting guide.

mluis1, 2018-01-16:
Please refer to the project Doxygen documentation [Porting Guide](http://stackforce.github.io/LoRaMac-doc/_p_o_r_t_i_n_g__g_u_i_d_e.html)

closed 2018-01-16 by mluis1