Lora-net/LoRaMac-node #270: SetTxContinuousWave

fahad80, 2017-07-13:
How exactly 'SetTxContinuousWave'  (in LoRaMac.c) works? It looks like this function just sets up the radio for continuous transmission but what data it will send? Don't we need to call Radio.send() (or maybe using DIO2)? I don't see any of those lines, so I am really confused.

mluis1, 2017-07-25:
A continuous wave doesn't send anything.

The goal is to set the radio to output the fundamental frequency (unmodulated).
It is mainly used when going for ETSI and FCC certification.

There is no need to call Radio.Send or checking radio interrupts line.

fahad80, 2017-07-25:
Aaaa...got it. Thanks for the clarification.

closed 2017-07-25 by fahad80

Kralg, 2018-03-26:
Hello, 
I am using B-L072Z with STM32CubeExpansion_LRWAN_V1.1.4 I can see that the code is very close to this implementation. I am testing Continuous wave mode and I can see that bandwidth in spectrum analyzer is much more narrow than 125kHz. I checked Tx ContinuousWave for SX1276 and they both seems to be the same in LoRaMac-Node and in cubeExpansion. Can someone elaborate on why this can happen? I can see low datarate optimization option which is suspicious but from the datasheet it seems to be fine.

```
void SX1276SetTxContinuousWave( uint32_t freq, int8_t power, uint16_t time )
{
    uint32_t timeout = ( uint32_t )( time * 1000 );

    SX1276SetChannel( freq );

    SX1276SetTxConfig( MODEM_FSK, power, 0, 0, 4800, 0, 5, false, false, 0, 0, 0, timeout );

    SX1276Write( REG_PACKETCONFIG2, ( SX1276Read( REG_PACKETCONFIG2 ) & RF_PACKETCONFIG2_DATAMODE_MASK ) );
    // Disable radio interrupts
    SX1276Write( REG_DIOMAPPING1, RF_DIOMAPPING1_DIO0_11 | RF_DIOMAPPING1_DIO1_11 );
    SX1276Write( REG_DIOMAPPING2, RF_DIOMAPPING2_DIO4_10 | RF_DIOMAPPING2_DIO5_10 );

    TimerSetValue( &TxTimeoutTimer, timeout );

    SX1276.Settings.State = RF_TX_RUNNING;
    TimerStart( &TxTimeoutTimer );
    SX1276SetOpMode( RF_OPMODE_TRANSMITTER );
    
}
```

nestorayuso, 2018-03-26:
this function is for unmodulated wave signal, if you don't modulate you can't measure the modulation bandwidth.

you need something like  SX1276SetTxConfig( MODEM_LORA,.....);

Kralg, 2018-03-26:
Thank you, I was looking to that part for quite long but have not noticed the very beginning...
I changed to MODEM_LORA. I saw that it was frequency hopping but for very short time and then the pause of ~1minute. Do you know what could be the reason for this? If I send 01010101 to 224 (certification mode) I can see packets coming from end-device. But once I issue the command 07000A84724014  I can see freq hopping but this seems to be just a single packet and after that a pause of 1min. After that cerfication mode packets continuous.

Kralg, 2018-03-27:
Can you reopen this issue? Or should I create a new one? I cant see how can I reopen it. I think that one of the problem is