Lora-net/LoRaMac-node #274: Retries of OTAA Join

roordaj, 2017-07-17:
Hi,

I'm having issues with the join request. This is the situation:

When my device tries to join the LoRa network ( fsm state DEVICE_STATE_JOIN ) the device tries to join the LoRa network 2 times. When both tries fail, the LoRaMAC layer does not retry to join the LoRa network again and the fsm goes into the DEVICE_STATE_SLEEP state. 

When I try to wake up the fsm manually by calling the function RequestSendPacket() the device tries a new join request ( LoRaMacMlmeRequest( ) ), but that function returns LORAMAC_STATUS_BUSY. It looks like the LoRaMAC layer is frozen and can't be used anymore. 

Is this a known issue, is there a work around? The ideal case in my opinion is this: The device tries to join 5 times in a row. When this fails, wait 30 minutes and try again and so on.

Thanks in advance,
J Roorda.

djaeckle, 2017-07-21:
Hi roordaj,

could you please provide some more details about your environment. Which version and region are you using?

roordaj, 2017-07-25:
Hi,

I'm using the latest version (4.3.2), and I ported it to the Nordic NRF52 microcontroller. I'm in Europe, so using the 868MHz band. I think that the problem lies in the LoRaMAC layer, but at this point, I don't know what the functionality of the OTAA join should be as there is not a functional design of the whole LoRaMAC-node project. Can anyone tell me what the expected behaviour is of a (failing) OTAA join request?

batulzii, 2017-07-26:
@roordaj  please,share your code ported to NRF52. i try to amalyze it, because i also worked on RRF52 sometimes ago. do you use  softdevice at the same time?

pdemil, 2017-07-26:
@roordaj You can request the LoRaWAN specification ( https://www.lora-alliance.org/lorawan-for-developers ) or, if you want to discuss on the phone, let me know.

How long did you wait after the first OTAA join request trials?

roordaj, 2017-07-27:
@batulzii I use a softdevice at the same time, but the way I made the two protocols work together is IP, so I don't want to share my code with you, I'm sorry for that. 

@pdemil It would be nice to discuss this issue on the phone. I waited for 10 minutes or so after the first OTAA join request trials. I tried looking up in the code if a timer was started for the next set of OTAA join request trials, but I was not able to find anything like that. Maybe you know more about the implementation of the OTAA join request?

batulzii, 2017-07-28:
@roordaj  my device running. it is for your issue only.

roordaj, 2017-07-28:
@batulzii Thats great, thank you so much! Let me know when you find something.

mluis1, 2018-01-16:
No more activity we close the issue.

closed 2018-01-16 by mluis1