Lora-net/LoRaMac-node #277: Schedule transmission of frame

fsessegoinfomob, 2017-07-18:
My node is having issues sendig on the network, I set TX next packet 

/*!
 * Defines the application data transmission duty cycle. 5s, value in [ms].
 */
#define APP_TX_DUTYCYCLE                            5000

The ISR fires but the node sendsonly when the Delayed timer fires.

I debugged the issue and i seen depending on ScheduleTx() function:
Into this function at the end there is the following snippet of code:

// Schedule transmission of frame
	if( dutyCycleTimeOff == 0 )
	{
		// Try to send now
		return SendFrameOnChannel( Channels[Channel] );
	}
	else
	{
		// Send later - prepare timer
		LoRaMacState |= LORAMAC_TX_DELAYED;
		TxDelayed_timeout_ms = dutyCycleTimeOff;
		TimerStart_TxDelayed(TxDelayed_timeout_ms);

		return LORAMAC_STATUS_OK;
	}

The dutyCycleTimeOff  is a number very high (for example 143 seconds): the setting is processed by function SetNextChannel() 

 0> INFO - RADIO ( Func: SetNextChannel ) , AggregatedTimeOff = 0 , AggregatedLastTxDoneTime = 11971 , TimerGetElapsedTime( AggregatedLastTxDoneTime )  = 3196
 0> INFO - RADIO ( Func: SetNextChannel ) , nbEnabledChannels = 0 , delayTx = 3 , nextTxDelay  = 143620

Why there is the strategy of "TxDelayed" that translates my sending to 143s instead of 5 seconds?

If i force the sending put  if(1)  (see below) the node sends always every 5 seconds
        // Schedule transmission of frame
 	//if( dutyCycleTimeOff == 0 )
        if( 1)
	{
		// Try to send now
		return SendFrameOnChannel( Channels[Channel] );
	}

Is it correct my AggregatedTimeOff is always 0 (it depend on AggregatedDCycle=1 always -->  
// Update Aggregated Time OFF
AggregatedTimeOff = AggregatedTimeOff + ( TxTimeOnAir * AggregatedDCycle - TxTimeOnAir );

Thanks

djaeckle, 2017-07-20:
Hi fsessegoinfomob,

thanks for the report. Which region are you using? Please note that even if you have define `APP_TX_DUTYCYCLE 5000`, the LoRaMAC will still respect the duty cycle limitation for the different regions. 

```
// Schedule transmission of frame
if( dutyCycleTimeOff == 0 )
{
  // Try to send now
  return SendFrameOnChannel( Channels[Channel] );
}
else
{
  // Send later - prepare timer         <-- In this case we queue the frame due to the duty cycle
  LoRaMacState |= LORAMAC_TX_DELAYED;
  TxDelayed_timeout_ms = dutyCycleTimeOff;
  TimerStart_TxDelayed(TxDelayed_timeout_ms);
  return LORAMAC_STATUS_OK;
}
```
E.g. the region EU868 comes with a duty cycle limitation. If the duty cycle is not elapsed and you want to send a frame, the LoRaMAC will queue the frame. When the duty cycle is elapsed, the LoRaMAC will send the frame.

fsessegoinfomob, 2017-07-20:
Hi,
where is the setting EU868 ?
I should found one #define into the source file?
Or it depends on Kerlink gateway settings?

djaeckle, 2017-07-21:
Hi,

in the sample projects, the definition for `REGION_EU868` is available in the preprocessor options of the IDEs. It doesn't depend on any gateway settings.

fsessegoinfomob, 2017-07-21:
Hi,
in the preprocessor option the only macro depend on radio is : USE_BAND_868
This is the region definition?

thanks

djaeckle, 2017-07-21:
Hi,

which version are you using? What do you want to do exactly? Do you want to turn off the duty cylce or change it?

fsessegoinfomob, 2017-07-21:
Hi,

my version is release/v4.3.2

For my test i need to send message every 1 minute.

If i set the duty cycle every 5seconds
efines the application data transmission duty cycle. 5s, value in [ms].
*/
#define APP_TX_DUTYCYCLE 5000

I checked the code

// Schedule next packet transmission
TxNextPacket_timeout_ms = TxDutyCycleTime;
TimerStart_TxNextPacket(TxNextPacket_timeout_ms);

TxDutyCycleTime is set between 4seconds to 6 seconds (random) but the sending doesn't start every 5 seconds because the strategy TimerStart_TxDelayed starts.

I would send every 5 seconds ( or every 60 seconds) so my sending is managed by TimerStart_TxNextPacket(TxNextPacket_timeout_ms) and not by TimerStart_TxDelayed starts

Thanks

djaeckle, 2017-07-21:
Hi  fsessegoinfomob,

thanks for the feedback. The definition `APP_TX_DUTYCYCLE` only specifies the application duty cylce. This means the application *tries* to send a frame every  `APP_TX_DUTYCYCLE`. However, the application duty cylce is independent from the LoRaMAC duty cycle. The LoRaMAC duty cycle respects the region specific regularities - the application duty cycle doesn't respect them.

In order to send every minute, you may do one of the following:

1. Add more channels on different bands. Duty cycle is related to the bands.
2. Turn off the duty cycle `LORAWAN_DUTYCYCLE_ON` (for test purposes only - the node will not be compliant to the EU868 regional regulations)

fsessegoinfomob, 2017-07-21:
Ok.

the point 2 works.

About the point 1, i found into code
			LoRaMacChannelAdd( 3, ( ChannelParams_t )LC4 );
			LoRaMacChannelAdd( 4, ( ChannelParams_t )LC5 );
			LoRaMacChannelAdd( 5, ( ChannelParams_t )LC6 );
			LoRaMacChannelAdd( 6, ( ChannelParams_t )LC7 );
			LoRaMacChannelAdd( 7, ( ChannelParams_t )LC8 );
			LoRaMacChannelAdd( 8, ( ChannelParams_t )LC9 );
			LoRaMacChannelAdd( 9, ( ChannelParams_t )LC10 );
You suggest to add other channels like
LoRaMacChannelAdd( 10, ( ChannelParams_t )LC11 );

How many channels cann i add?

thank you  very much

djaeckle, 2017-08-08:
Hi fsessegoinfomob,

the channel definition depends on the region. Please take a look at the LoRaWAN Specification. For the EU868 region, 16 channels are available. The first three are the default channels.

closed 2018-01-16 by mluis1