Lora-net/LoRaMac-node #279: Unconfimed RxTimeout

fahad80, 2017-07-25:
Hi,

Shouldn't RxTimeOut generate during CONFIRMED data sending? As opposed to UNCONFIRMED data? I receive DIO1 interrupt no matter what type(CONFIRMED/UNCONFIRMED) of data I send. Is that how it should be?

What if I don't want DIO1 to generate an interrupt after an UNCONFIRMED data send?

reissjason, 2017-07-26:
In class A operation it is not known if the Server will send a downlink. The RX windows are opened after each uplink to allow a downlink packet to be received.

jvpernis, 2017-07-26:
The LoRaWAN specification describes that there is a receive window 2 seconds after transmitting, in which one is able to receive a response from the gateway/server. 
A CONFIRMED message is guaranteed to generate a response message, as it contains an acknowledgement and otherwise something went wrong with transmitting the message.
With an UNCONFIRMED message it depends on the user application whether the node receives data from the gateway/server.

So yeah, this is how it should be

fahad80, 2017-07-26:
@reissjason , @jvpernis : Thanks for clarification. 👍

closed 2017-07-26 by fahad80