Lora-net/LoRaMac-node #280: im880b UART Issue

RakshitAdmar, 2017-07-26:
I'm having trouble with getting the uart to work with this driver. 
I've managed to get HAL_UART_Transmit() to work but using the  UartPutChar only produces junk value. 
Here's how I'm configuring 

`UartInit( &Uart1, 0, UART_TX, UART_RX );

    FifoInit( &txFifo, txBuffer, 2);
    FifoInit( &txFifo, txBuffer, 2);
    Uart1.FifoTx = txFifo;
    Uart1.FifoRx = rxFifo;

    UartConfig( &Uart1, RX_TX, 115200, UART_8_BIT, UART_1_STOP_BIT, NO_PARITY, NO_FLOW_CTRL );`

Is there something I'm missing? I've also tried most of the fixes recommended in the other issues.

jvpernis, 2017-07-26:
`UartInit( &Uart1, 0, UART_TX, UART_RX );`

Don't you need to pass an uart id in the second parameter of [UartInit](https://github.com/Lora-net/LoRaMac-node/blob/master/src/system/uart.c#L29)?

E.g.
`UartInit( &Uart1, UART_1, UART_TX, UART_RX );`

Or, if you are re-initializing:
`UartInit( &Uart1, Uart1.UartId, UART_TX, UART_RX );`

Otherwise, try answering (some of) these questions:

- What do you mean when it produces junk values? 

 - Do you receive the UART signal in a terminal? Does the configured baud rate match with the one used in your terminal?

- How did you configure HAL_UART to get it to work? How is it different from 'this driver'?

- Do you have any other code to show for 'this driver'?

Also, you can init the uart fifo object directly by passing a pointer to its fifo object:
`FifoInit( &Uart->FifoRx, rxBuffer, 2);`
`FifoInit( &Uart->FifoTx, txBuffer, 2);`

RakshitAdmar, 2017-07-26:
From uart-board.c 
`void UartMcuInit( Uart_t *obj, uint8_t uartId, PinNames tx, PinNames rx )`
says that uartId is uint8_t. Irrespective of which, the enum UART_1 still corresponds to 0.

1. By junk values, I mean, if  I'm sending 'x'=120 decimal, I receive 80. 
2. I used STM32Cube to generate Uart initialization code. Generated code is part of Stm's HAL_MSP drivers. 
3. I do have code, you can obtain the same from CubeMX
**From stm32l1xx_hal_msp.c**

`void HAL_UART_MspInit(UART_HandleTypeDef* huart)
{

  GPIO_InitTypeDef GPIO_InitStruct;
  if(huart->Instance==USART1)
  {
  /* USER CODE BEGIN USART1_MspInit 0 */

  /* USER CODE END USART1_MspInit 0 */
    /* Peripheral clock enable */
    __HAL_RCC_USART1_CLK_ENABLE();
  
    /**USART1 GPIO Configuration    
    PA9     ------> USART1_TX
    PA10     ------> USART1_RX 
    */
    GPIO_InitStruct.Pin = GPIO_PIN_9|GPIO_PIN_10;
    GPIO_InitStruct.Mode = GPIO_MODE_AF_PP;
    GPIO_InitStruct.Pull = GPIO_PULLUP;
    GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_VERY_HIGH;
    GPIO_InitStruct.Alternate = GPIO_AF7_USART1;
    HAL_GPIO_Init(GPIOA, &GPIO_InitStruct);

    /* USART1 interrupt Init */
    HAL_NVIC_SetPriority(USART1_IRQn, 0, 0);
    HAL_NVIC_EnableIRQ(USART1_IRQn);
  /* USER CODE BEGIN USART1_MspInit 1 */

  /* USER CODE END USART1_MspInit 1 */
  }

}`

From main()
`
/* USART1 init function */
static void MX_USART1_UART_Init(void)
{

  huart1.Instance = USART1;
  huart1.Init.BaudRate = 115200;
  huart1.Init.WordLength = UART_WORDLENGTH_8B;
  huart1.Init.StopBits = UART_STOPBITS_1;
  huart1.Init.Parity = UART_PARITY_NONE;
  huart1.Init.Mode = UART_MODE_TX_RX;
  huart1.Init.HwFlowCtl = UART_HWCONTROL_NONE;
  huart1.Init.OverSampling = UART_OVERSAMPLING_16;
  if (HAL_UART_Init(&huart1) != HAL_OK)
  {
    _Error_Handler(__FILE__, __LINE__);
  }

}

/** Pinout Configuration
*/
static void MX_GPIO_Init(void)
{

  /* GPIO Ports Clock Enable */
  __HAL_RCC_GPIOA_CLK_ENABLE();

}`

4. Thanks for showing me the right way to initialize the fifo. 

I did try to bypass the fifo by directly calling HAL_UART_Transmit_IT() and HAL_UART_Transmit()
Both still produced garbage.

jvpernis, 2017-07-26:
>  Irrespective of which, the enum UART_1 still corresponds to 0.

Good point, haven't thought of that. For readability of your code though, it might be wise still to replace it with the enum value. 

> if I'm sending 'x'=120 decimal, I receive 80.

I assume this isn't the simple mistake of sending 128 (~120) and receiving 0x80, right? :smile: 

> I used STM32Cube to generate Uart initialization code. Generated code is part of Stm's HAL_MSP drivers.

The only two differences I was able to spot was the pin frequency in GpioMcuInit and its interrupt priority. Though I'm not sure if it matters.

----

[GpioMcuInit](https://github.com/Lora-net/LoRaMac-node/blob/master/src/boards/LoRaMote/gpio-board.c#L61) (LoRaMote implementation, but should be equal to all boards):
`GPIO_InitStructure.Speed = GPIO_SPEED_FREQ_HIGH;`

Your code:
`GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_VERY_HIGH;`

----

[UartMcuConfig](https://github.com/Lora-net/LoRaMac-node/blob/master/src/boards/LoRaMote/uart-board.c#L126) (LoRaMote implementation, but should be equal to all boards):
`HAL_NVIC_SetPriority( USART1_IRQn, 8, 0 );`

Your code:
`HAL_NVIC_SetPriority(USART1_IRQn, 0, 0);`

----

Are you using this code in the same project as the LoRaMac-node? Or do you use a project file generated by CubeMX?
If I'm not mistaken, IM880B has a different/updated microcontroller compared to the previous IM880A, which the LoRaMac-node project is based on. (STM32L151CB-A instead of STM32L151CB. It's a small difference, and it should not cause any problems, but still..)

Is it possible for the compiler settings to differ?

###### P.S. I also noticed you initialized the txFifo twice, and rxFifo none, which could just be a mistake while copying the code to your post.

RakshitAdmar, 2017-07-27:
Oops. I had a conflicting file defining the mcu frequency. 
In files

```
LoRaMac-node/src/boards/SK-iM880A/cmsis/stm32l1xx_hal_conf.h

```

and

```
 LoRaMac-node/src/boards/mcu/stm32/STM32L1xx_HAL_Driver/Inc/stm32l1xx_hal_conf_template.h

```

For some reason, my compiler picked up the definition from the template file first. 

Thank a lot for your time @jvpernis 
Much appreciated. No other changes to the driver libraries were required. 
But now I get the characters I send followed by redundant 0's in hex.

closed 2017-07-27 by RakshitAdmar