Lora-net/LoRaMac-node #281: Current Channel mask is clobbered during ADR and JOIN procedures

jamesl-dm, 2017-07-27:
Hi,

In the US915 and AU915 AlternateDr functions, the current Channel mask is overwritten to enable high speed channels:

```
int8_t RegionAU915AlternateDr( AlternateDrParams_t* alternateDr )
{
    ...
    // Re-enable 500 kHz default channels
    ChannelsMask[4] = 0x00FF;
    ...
}
```

The Channel mask is also clobbered when using ADR to switch down to the lowest datarate:
```
bool RegionAU915AdrNext( AdrNextParams_t* adrNext, int8_t* drOut, int8_t* txPowOut, uint32_t* adrAckCounter )
{
    ...
    if( adrNext->AdrEnabled == true )
    {
        if( datarate == AU915_TX_MIN_DATARATE )
        {
            ...
        }
        else
        {
            ...
            if( adrNext->AdrAckCounter >= ( AU915_ADR_ACK_LIMIT + AU915_ADR_ACK_DELAY ) )
            {
                if( ( adrNext->AdrAckCounter % AU915_ADR_ACK_DELAY ) == 1 )
                {
                    // Decrease the datarate
                    getPhy.Attribute = PHY_NEXT_LOWER_TX_DR;
                    getPhy.Datarate = datarate;
                    getPhy.UplinkDwellTime = adrNext->UplinkDwellTime;
                    phyParam = RegionAU915GetPhyParam( &getPhy );
                    datarate = phyParam.Value;

                    if( datarate == AU915_TX_MIN_DATARATE )
                    {
                        // We must set adrAckReq to false as soon as we reach the lowest datarate
                        adrAckReq = false;
                        if( adrNext->UpdateChanMask == true )
                        {
                            // Re-enable default channels
                            ChannelsMask[0] = 0xFFFF;
                            ChannelsMask[1] = 0xFFFF;
                            ChannelsMask[2] = 0xFFFF;
                            ChannelsMask[3] = 0xFFFF;
                            ChannelsMask[4] = 0x00FF;
                            ChannelsMask[5] = 0x0000;
                        }
                    }
                }
            }
        }
    }
    ...
}
```

Shouldn't the stack be using the ChannelsDefaultMask as set by LoRaMacMibSetRequestConfirm / MIB_CHANNELS_DEFAULT_MASK rather than hardcoded defaults?

It uses the ChannelsDefaultMask when a confirmed uplink fails its retries (via RegionInitDefaults( LoRaMacRegion, INIT_TYPE_RESTORE )), so it seems inconsistent to follow a different procedure when the ADR fails. I believe the code was originally adapted from the EU868 style regions, where it is appropriate to re-enable the 3 default channels that everyone must support. But in the 915 regions there are no baseline defaults, and forcing all the channels on causes massive packet loss.

djaeckle, 2017-08-04:
Hi jamesl-dm,

thanks for your report. We understand your concerns, but the current implementation is according to the LoRaWAN Specification. You could write a change request to the technical commitee of the LoRaWAN Alliance help@lora-alliance.org.

jamesl-dm, 2017-08-07:
Are you sure? I've read through the 1.0.2 July 2016 specification and the v1.0.2rB regional parameters several times, and I can't find any specifications or recommendations about changing the channel masks except in response to the CFList, LinkADRReq, or XxxChannelReq's.

The two network server implementations I have available to test against only set the channel mask once per join, using a LinkADRReq. If the end-node subsequently clobbers the channel mask during an ADR backoff, I end up with 90% packet loss. So ADR isn't really usable for me as it stands.

djaeckle, 2017-08-07:
Hi jamesl-dm,

thanks for the feedback. I think your comment is correct, it is not mentioned explicitly. However, it is available in Specification v1.1. I will discuss this issue with my collegues. I will come back to you as soon as possible.

djaeckle, 2017-09-06:
Hi jamesl-dm,

we have discussed the issue and we think that the implementation is correct.

jamesl-dm, 2017-09-07:
>As previously stated you can always send a change request to the LoRa Allinace technical committee.
Ok, I will do that. Will leave the issue open for now, as it's a pretty serious flaw.

jamesl-dm, 2017-09-27:
I've finally managed to get a hold of the 1.1 spec, but no word back from the technical committee yet.

I see that the draft standard simply says the channel mask must be reset to default once the ADR procedure has backed off all the way to maximum power and lowest datarate without any success. It makes no distinction between 'the default channels', and any other channel mask state on the device. So I'm wondering what role the DefaultChannelMask variable plays in this?

I would have thought that the whole purpose of having a separate DefaultChannelMask variable in the regions is to encode the specification defined default channel mask, in one place. So it's not clear why it is hardcoded in some places, and uses the variable in others. Or what the objection is to allowing the user to override it if they think they know better than the spec. It wouldn't cause the MAC to fail certification, because you wouldn't override the DefaultChannelMask if seeking certification. But as the code stands, you are able to override it, but it isn't clear what it does.

I also can't see any reference to resetting the channel mask when a confirmed uplink's retry limit expires (as is done using RegionInitDefaults( LoRaMacRegion, INIT_TYPE_RESTORE ), in OnMacStateCheckTimerEvent). So presumably this behavior doesn't comply with the standard?

jamesl-dm, 2017-11-24:
Further to this, this channel mask re-enable in RegionUS915HybridAdrNext looks especially broken:

```if( datarate == US915_HYBRID_TX_MIN_DATARATE )
                    {
                        // We must set adrAckReq to false as soon as we reach the lowest datarate
                        adrAckReq = false;
                        if( adrNext->UpdateChanMask == true )
                        {
                            // Re-enable default channels
                            ReenableChannels( ChannelsMask[4], ChannelsMask );
                        }
                    }
```

I think the right thing to do here (and in the other regions) is call RegionUS915HybridInitDefaults(INIT_TYPE_RESTORE), which will re-enable the channels correctly (using the default mask, rather than changing the ChannelMask based on _itself_:

```
ReenableChannels( ChannelsDefaultMask[4], ChannelsMask );

for( uint8_t i = 0; i < 6; i++ )
{ // Copy-And the channels mask
    ChannelsMaskRemaining[i] &= ChannelsMask[i];
}
```

Note its ReenableChannels( **ChannelsDefaultMask**[4], ChannelsMask ), rather than ReenableChannels( **ChannelsMask**[4], ChannelsMask ).

mluis1, 2018-01-16:
The specification is not very clear on this. The feedback that we got from the Alliance is that the node must re-enable all the channels.

We know that there are discussions being held at the Alliance concerning this topic. Up until they don't provide clear guidelines on this we will keep the implementation as is.

The other reason to keep it is because the Certification process also expects the node to re-enable all the channels in those conditions.

Therefore if you don't mind we close the issue.

closed 2018-01-16 by mluis1