Lora-net/LoRaMac-node #282: CalculateBackOff is called twice for delayed transmissions?

jamesl-dm, 2017-07-27:
In the development branch, ScheduleTx calls CalculateBackOff each time a transmission is scheduled:

```
static LoRaMacStatus_t ScheduleTx( void )
{
    ...
    // Update Backoff
    CalculateBackOff( LastTxChannel );
    ...
    // Schedule transmission of frame
    if( dutyCycleTimeOff == 0 )
    {
        // Try to send now
        return SendFrameOnChannel( Channel );
    }
    else
    {
        // Send later - prepare timer
        LoRaMacState |= LORAMAC_TX_DELAYED;
        TimerSetValue( &TxDelayedTimer, dutyCycleTimeOff );
        TimerStart( &TxDelayedTimer );

        return LORAMAC_STATUS_OK;
    }
}
```

It looks like if there is no available channel and the transmission is delayed, ScheduleTx will be called a second time. CalculateBackoff would then be called a second time, resulting in AggregatedTimeOff being incremented twice instead of once?

jamesl-dm, 2017-07-31:
After further examination of the MAC, it looks to me like ScheduleTx is designed to be called multiple times, and the real issue is that some of the functions it calls aren't designed this way.

- The update of AggregatedTimeOff in CalculateBackoff should probably be an assignment rather than an increment.
- Calling RegionNextChannel shouldn't result in the RemainingChannelMask variables in the AU915 and US915(-hybrid) regions being modified to mark the channel as _used_. They can be modified in their TxConfig functions instead, once the stack is certain that it means to use the channel immediately.

djaeckle, 2017-08-04:
Hi jamesl-dm,

we will analyze your issue and will come back to you.

djaeckle, 2017-09-06:
Hi jamesl-dm,

> The update of AggregatedTimeOff in CalculateBackoff should probably be an assignment rather than an increment.

I think this is an issue. We will provide as soon as possible.

djaeckle, 2017-10-09:
Hi jamesl-dm,

this issue is on the roadmap for the next release:

https://github.com/Lora-net/LoRaMac-node/wiki/LoRaMAC-node-Roadmap

djaeckle, 2018-01-24:
Hi jamesl-dm,

> Calling RegionNextChannel shouldn't result in the RemainingChannelMask variables in the AU915 and US915(-hybrid) regions being modified to mark the channel as used. They can be modified in their TxConfig functions instead, once the stack is certain that it means to use the channel immediately.

I think we set them as used. We do only mark a channel as used, if we have found a valid channel. If we have found a valid channel, the MAC will transmit a frame directly afterwards.

jamesl-dm, 2018-01-25:
Ok - I must have misread it. Thanks.

closed 2018-01-26 by mluis1