Lora-net/LoRaMac-node #283: class c & rx2 & crc bad

leewonsoo, 2017-07-27:
Hi,
The node of CLASS_C can't receives a downlink frame after it receives a frame with an invalid CRC in RX2. 
That is recovered after receives any frame except an invalid crc.

djaeckle, 2017-08-04:
Hi leewonsoo,

could you provide more detailed information? I'm not sure if I understand your issue correctly.

leewonsoo, 2017-08-04:
Hi.
I will test it again next week and post it again.

leewonsoo, 2017-08-07:
Hi.
I tested again today.
The result is that the next downlink frame is always broken after receiving the rxerror.
For example, 
The following is the downlink frame before decryption, the devaddr is 0x07565AB3.

* in gateway tx log
> a0 b3 5a 56 07 80 0e 00 01 f9 33 52 8c c9 19 ae 47 2a fb    <- -fcnt = 0x000e
>  .....
> a0 b3 5a 56 07 80 0f 00 01 4c 8f 60 99 7a ae 9a 4a ba a8      <-- fcnt = 0x000f

* in end device, OnRadioRxDone() log.
> A0 B3 5A 56 07 80 0E 00 01 F9 33 52 8C C9 19 AE 47 2A FB   <--fcnt = 0x000e 
> ... rxerror event ...
> 50 2E F7 0C 52 68 06 0C 96 88 A6 08 A0 B3 5A 56 07 80 0F    <-- broken frame....

* 
I don't know why. 
Is it a bug in the sx1276 chip?

Anyway if the source code modified as follows, it's not broken.

```

static void OnRadioRxError ( void )
{
#if 0 // 2017.08.07
    if(LoRaMacDeviceClass! = CLASS_C)
    {
        Radio.Sleep ();
    }
    else
    {
        OnRxWindow2TimerEvent ();
    }
#else
    Radio.Sleep ();
#endif
    if( RxSlot == 0 )
    {
        if( NodeAckRequested == true )
        {
            McpsConfirm.Status = LORAMAC_EVENT_INFO_STATUS_RX1_ERROR;
        }
        MlmeConfirm.Status = LORAMAC_EVENT_INFO_STATUS_RX1_ERROR;

        if( TimerGetElapsedTime( AggregatedLastTxDoneTime ) >= RxWindow2Delay )
        {
            LoRaMacFlags.Bits.MacDone = 1;
        }
    }
    else
    {
        if( NodeAckRequested == true )
        {
            McpsConfirm.Status = LORAMAC_EVENT_INFO_STATUS_RX2_ERROR;
        }
        MlmeConfirm.Status = LORAMAC_EVENT_INFO_STATUS_RX2_ERROR;
        LoRaMacFlags.Bits.MacDone = 1;
    }
#if 1 // 2017.08.07
    if( LoRaMacDeviceClass == CLASS_C )
    {
        OnRxWindow2TimerEvent( );
    }    
#endif
}
```

djaeckle, 2017-08-07:
Hi leewonsoo,

I need some time for the analysis. I will keep you up-to-date.

djaeckle, 2018-01-24:
Hi leewonsoo,

sorry for the late response. So far i couldn't reproduce the issue. Is this still an issue on your side? Do you perform an uplink between the RxError and the Rx where the payload is corrupted?

leewonsoo, 2018-01-25:
Hi djaeckle,
I am using the modified source. So i am not testing. 
An uplink is ok because this calls Radio.Sleep ().

mluis1, 2018-01-26:
As we can't reproduce this issue we decided to close it. In case you see it again please feel free to re-open the issue.

closed 2018-01-26 by mluis1