Lora-net/LoRaMac-node #284: Random seed generation doesn't work?

jamesl-dm, 2017-07-28:
In LoRaMacInitialization, just after calling Radio.Init, Radio.Random is called to seed the random number generator. I noticed that on my board (which is a Murata module with an SX1276), the random number is usually 0 or 0xFFFFFFE0.

The issue seems to be that Radio.Init ends with a SX1276SetModem( MODEM_FSK ), and Radio.Random begins with a SX1276SetModem( MODEM_LORA ). If you insert a delay between them (1 ms works), or remove the SX1276SetModem( MODEM_FSK ), the random number works as expected.

djaeckle, 2017-08-08:
Hi jamesl-dm,

thanks for the report. I double checked it on my side and I cannot confirm the issue.

jamesl-dm, 2017-08-08:
Ok thanks. I'm guessing its a timing issue that only shows up with very tight SPI timing. Had lots of similar RSSI timing issues with Silabs radios in the past.

closed 2017-08-08 by jamesl-dm